autosportlabs/RaceCapture-Pro_firmware #39: Lua script issues

Pilt2013, 2015-04-29:
addChannel function: Using over 11 characters for channel name causes unwanted results:

From lua script: ```ABS_Bpressure_Id = addChannel("ABS_Br_Press", 10, 0, -42.5, 425, 'bar')```

Results in logfile output of  "ABS_Br_Pressbar"|"bar"|0|0|10

Also Racecapture v2.7.9 requires a power cycle after uploading a new lua script, otherwise still runs old script.

stieg, 2015-04-29:
Thanks for the info @Pilt2013.  Yes the firmware currently only supports up to 11 characters (12 with the null termination) for a channel name.  Looks like we need to better enforce that on the method.  However the best way I can think of is simply drop all characters after the 11th.  Perhaps consider using "Brake_Press" or "ABS_Press" as a work around.

stieg, 2015-04-29:
Ok.  Feedback given.  Sorry for spreading it out between patches and the pull  req.

brentpicasso, 2015-04-30:
Fixed in https://github.com/autosportlabs/RaceCapture-Pro_firmware/pull/40   will show up in 2.8.0

closed 2015-04-30 by brentpicasso