autosportlabs/RaceCapture-Pro_firmware #43: Occasional bug in logging

Pilt2013, 2015-05-01:
Occasionally in the log one of the first 10 or so lines contains no data (just commas).
It appears to be easier to reproduce if you stop logging (by pressing the button) then soon after press the button to start a new log file. In this case it appears to miss a line or two in the new file.

Maybe the logger needs to check it has flushed the old file before creating a new one?

Please see here for a load of examples 

https://drive.google.com/folderview?id=0B_AuKfSEq7r7flpGYmpINGxBRUx2Z0ZEMXBLV0twTW52QlJXekhVMlBLY1hmSEVkZ3dhUjQ&usp=sharing

Connected to RCP via USB at time as well on desk powered from USB.

stieg, 2015-05-01:
Hey @Pilt2013,  What version of the firmware are you running?  Also have you been running any version of the firmware other than what has been officially released?  TIA.

Pilt2013, 2015-05-01:
V2.7.9 official release on RCP v2

stieg, 2015-05-01:
Ok.  Do you have an RCP config you could upload to that google drive?  Also do you see the issue repeating itself at regular issues or is it an isolated occurrence?

Pilt2013, 2015-05-05:
Hi,

Had no internet this weekend. Added config to drive and lua script i am using. Can get the issue to occur by inserting the sdcard, then pressing the button within 5 seconds. Then the logfile contains a blank line see RCP_0 and RCP_1 (1 was done straight away after and is ok)

brentpicasso, 2015-05-05:
Worth noting: The SD card upon first power up has a longer initialization
process for the first log file vs the 2nd. this could be a hint.

stieg, 2015-10-13:
This was fixed in 2.8.4.  Closing.

closed 2015-10-13 by stieg