autosportlabs/RaceCapture-Pro_firmware #108: Predicted timer seems to periodically freeze under certain conditions.

ryandoherty, 2015-10-26:


stieg, 2016-01-04:
Unfortunately not enough info to work with here.  Re-open if there is info + logs to work with.

brentpicasso, 2016-01-04:
This is observed frequently enough that we need to leave it open and get some example data.

ryandoherty, 2016-01-04:
I may have logs, along with Scott. Will dig.

stieg, 2016-05-23:
Finally observed this at Vodden the Hell 2016.  Timer jumps its value at the end of a lap.  This may be due to an log buffer bounds violation when doing the timing projection.  If this is the case, it could be displaying old lap data, causing the jump that users are reporting.  Julian Cordle from Snowspeeders reported the issue to me in 2.9.1 MK2 at the track and I saw it on MK1 2.8.7.

stieg, 2016-05-23:
Looking at the code the issues likely lies in `findTwoClosestPts` and `getSplitAgainstFastLap`.  It seems that if we can't find good values to use for the projection due to an overflow of data in the buffer then we could get values that could cause us to return a previous answer (hence a Freeze effect) or values that cause a bad calculation that could lead to invalid results.  Will continue digging here...

wcorredor, 2017-02-02:
This is still happening in the 2.10.2. I observed the same behavior at Sebring on Jan 23 2016. Lap prediction seems to be correct until the very end of the lap (about 300 yards from finish line) and suddenly it goes up by about 3 seconds until one crosses the start line when the correct time shows up.

brentpicasso, 2017-02-02:
Hi, thanks for the follow up. 

Do you have any log file data you can upload that catches this in the act? If we have the raw data we might be able to pump it back through the predictive timer to see if it reproduces. 

Thanks!

wcorredor, 2017-02-02:
Yes, I do have data. In the car at the moment.

We are travelling today to Atlanta for Chumpcar this weekend. I will send it to you as soon as I have some piece of mind.

I still owe you the E30 script; still doing some adjustments and testing before publishing it. It is great but it has gotten a bit large as I have added lots of sensors and information. I have struggled a lot with the amount of memory available for the script. I have reduced the variables to the bare minimum to use the stack as much as possible, but I am running out of juice and can’t add not even one new idea.

I hope this weekend I can finish “mpg”. It has proven harder that I thought and difficult to test while racing. As I posted a TotalDistante() function (that does not reset every lap) would help to reduce code and be more precise.

Regards

William

wcorredor, 2017-02-06:
Attached the log data file from Sebring. I just analyzed it and plotted PredTime vs ElapsedTime. The issue is very clear. It happens in almost every lap. But look for example at lap 11. Up untill the end of the straight away before the last turn, it predicted 2.88. Through the last turn and after it, it erratically predicts up to 2.91. Even a few yards before the finish line predicts 2.89 and then you will be able to see that real time was 2.88. So the original prediction was correct for most of the lap.
I just observed the same this weekend at Road Atlanta and it also happens at the very end of the lap. It predict 2 to 4 seconds higher.

[rc_192.zip](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/755895/rc_192.zip)

stieg, 2017-02-13:
Howdy @wcorredor,

   Thanks for the data.  Just spent the last hour analyzing the output and the track conditions to see if I can figure out what is happening.  Based on what I see this issue is GPS related.  Specifically in your case here I imagine all of your issues were happening as you approached the bridge that goes over the track at the final turn at Sebring.  You can see it in your data quite clearly if you enable the GPSDOP and GPSQual channels.  Their meanings are as follows:

- DOP - Positional Dilution of Precision. - https://en.wikipedia.org/wiki/Dilution_of_precision_(navigation)
- Qual - Gps Quality as perceived by the GPS chip.

Relevant meanings of DOP:
- 0 - 1 - Ideal precision
- 1 - 2 - Excellent precision.  Good enough to be reliably accurate
- 2 - 5 - Good precision, but not necessarily super accurate.
- 5+ - Useless to us.

Relevant meanings for GPS quality:
- 0 - No Fix
- 1 - 2D Fix
- 2 - 3D Fix
- 3 - 3D Differential Fix

For our system to be very accurate we rely heavily on differential GPS to deliver an accurate locational fix.  As you can see when you approach the bridge the DOP starts to raise from its more nominal 1.6-9 range and jumps up into the 2 range.  We also lose differential GPS at this time consistently (seen when GPS quality value drops from 3 to 2).  As a result our accuracy wanes and the effect cascades into blips with the predictive timer.

This is a very tricky one to fix since it has to do with the accuracy of GPS.  I am thinking my fix may involve placing GPS quality restrictions on what we use for reference points when recording predictive timing data and calculating these times.  That may help.

For you @wcorredor do you by chance have a picture of where the GPS antenna is located on the car.  To help alleviate this issue ensure the antenna is as far away from other sources of radio interference as is possible and that is has a *clear* view of the sky.

wcorredor, 2017-02-13:
I have since moved the GPS antenna to the roof of the car right in the center. I will let you know if it happens again. I will also check other data to see what DOP looked like other days and see if the behavior is there or not.
Thanks
William

wcorredor, 2017-02-14:
Hi @stieg

Attached another log file from Road Atlanta last weekend, the GPS antenna better positioned this time.
Compare the real lap time with the predicted at the very end of most laps and you will notice a difference of about 2 tenths of a sec. I don’t know if that is the best that can be, taking into account that we are sampling at 10hz. However, I also have a VBOX Lite on the car, also at 10hz, and it seems to have a more accurate predicted time. (They claim to use what they call GPS LineSnap).

In particular, take a look at the end of lap 13 and 17 both with and DOP < 0.4 and QUAL =3. The predicted time stops changing 2 and 8 seconds before the finish line respectively. At the very end both miss by 1.9 and 2.7 tenths of a second, which seems to be too much for this sport.

As a matter of coincidence lap 16 and 17 have the exact same time to 1 thousands of a second. Perhaps it may be of some help to compare the prediction which happens to be different by 3.4 tenths of a second.

Let me know if this helps and if you want additional data or analysis.
Regards

[RoadAtlanta.zip](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/773119/RoadAtlanta.zip)
wcorredor

RodADavison, 2020-09-08:
Jumping in here -- 3 years after the fact -- it looks like at one point you suspected GPS accuracy so you filtered the predictions with DOP value.  However, it doesn't appear that the fastLap against which it is compared uses filtered data.  Is it possible that the error is coming from the fastLap, not the current lap?

brentpicasso, 2020-09-08:
Interesting theory. I'm porting the same algorithm to the RC app for purposes of recalculating lap timing after the fact. Will be curious to see if the problem shows up there as well, and might provide an easier way to debug.

stieg, 2020-12-14:
So... after many years of being dormant I believe I may have found the culprit here.  Anyway I wrote a simulator to test this and I need full log files to validate that this issue is fixed.  If anyone here would be so kind as to upload these files it would help me out.

closed 2020-12-14 by stieg

wcorredor, 2020-12-15:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I don’t have recent logs, but I remember sending you several before. Can you test with those?
Wcorredor

stieg, 2020-12-15:
@wcorredor The logs above (rc_192.log) are incomplete.  They start at lap 7 and go through lap 12.  Since they are missing several laps I can not use them with the simulator to great effect since the simulator only sees part of the data and can not replicate it the same way.

wcorredor, 2020-12-15:
Got it. I will try to get another one but won’t be back in a track until January.

We were in Sebring a few weeks ago and the behavior was there. Right under the bridge getting to the start line the prediction would be about 3-4 seconds up and it would correct right at the line.

Thanks

wcorredor

CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

@wcorredor<https://github.com/wcorredor> The logs above (rc_192.log) are incomplete. They start at lap 7 and go through lap 12. Since they are missing several laps I can not use them with the simulator to great effect since the simulator only sees part of the data and can not replicate it the same way.

brentpicasso, 2020-12-15:
We can get this into an updated firmware beta release and make it available on Discord. Should be able to do it this week. 

-Brent

commit b34d101c (2020-12-14): fix: correct predictive timer interval calculation

We were dividing our slots value by 0.9 (90%) when we should have
been multiplying this value by 90%. We want to target usage of
90% of our slots and this is the correct math. Otherwise we will
constantly overflow and have poor results at the end of the lap
when we run out of buffer space.  This leads to poor measurements
and thus the jumps people were seeing.  *sigh*

Resolves: #108