autosportlabs/RaceCapture-Pro_firmware #123: BT disconnects with 2.8.0 and latest prod Android app

ryandoherty, 2015-10-26:
After 5 seconds of  RCP (mk2 and 2.8.0) sending data over BT it stops sending data.

Most likely due to the new HB logic.

Wireless: timeout
BT: Probing baud115200
BT: cmd: AT
ATBT: cmd rsp: 
BT: nomatch
BT: Provisionfail
BT: Probing baud9600
BT: cmd: AT
ATBT: cmd rsp: 
BT: nomatch
BT: Provisionfail
BT: Probin

stieg, 2015-11-16:
Looks like the serial altering issue I just fixed.  See https://github.com/autosportlabs/RaceCapture-Pro_firmware/pull/346

brentpicasso, 2015-11-16:
Double check this isn't related to RCP trying to provision the bt module
while android is already connected. If android is already connected, the
module will not respond to AT commands

Brent Picasso
Autosport Labs
Technology for Race and Street

stieg, 2015-11-16:
I hit this issue while implementing my changes.  Issue was that we were not
giving the serial port enough time to synchronize with the BT device before
we would send the AT command to check if we could communicate with the
hardware.  It was intermittent.  When I put a 15ms delay after changing the
serial port baud rate, the issue completely disappeared.

closed 2016-01-04 by stieg