autosportlabs/RaceCapture-Pro_firmware #124: Channel precision not saved to log file

ryandoherty, 2015-10-26:
Looks like a channel's precision is not in the meta info in log files. This would be useful to have for loading log files later and for test scripts that send data to RCL.

stieg, 2016-01-04:
@brentpicasso / @ryandoherty I don't understand how this info would be useful?  Can you elaborate?

brentpicasso, 2016-01-04:
It would represent the complete Channel configuration in the log file
metadata. Right now this value is missing from metadata, but min/max are
present, for instance.

stieg, 2016-01-04:
I understand that.  What I'm trying to grasp is why we need this?  Who would use it or find it valuable?

brentpicasso, 2016-01-04:
It's a hint for downstream systems. Like min/max is sort of a hint.

stieg, 2016-01-04:
I still don't see the value I'm afraid.  I see the value of min and max because we could use those to specify the low and high points of a gauge.  This would allow us to make full use of a gauge when rendering.  With precision I don't see the same value because its implicit in how the values are given.  If we send a float value that looks something like `1.23`, you can imply that the precision must be at least 2 (it could be higher).

Point here is that I don't feel we should send data that has no direct use.  If there is a direct use, then I'm for it.  Otherwise leave it out as it only would bloat the data stream we send back.

stieg, 2016-06-05:
closing due to no response.

closed 2016-06-05 by stieg