autosportlabs/RaceCapture-Pro_firmware #134: OBD-II - query PIDs with higher sample rates proportionately more often than PIDs with lower sample rates.

ryandoherty, 2015-10-26:
for example, RPM PID at 10Hz should be queried 10x more often than a coolant PID being requested at 1Hz

stieg, 2016-01-04:
@brentpicasso Is this still an issue?  What is current behavior if it is?

brentpicasso, 2016-01-04:
OBD-II PIDs are queried sequentially regardless of the actual desired
sample rate.  An algorithm to proportionally more frequently query PIDs
with faster sample rates would be beneficial.

E.g.

engineTemp at 1Hz
Tps at 10Hz

right now we query CAN PIDs at equal intervals as fast as possible, and
then separately sample the current values at the desired sample rates.

stieg, 2016-06-05:
Seems like this should be part of #172

stieg, 2016-07-10:
I'm thinking of punting this until we fix #172.  Rationale is that since we need to re-tool a major portion of the OBD-II system with CAN anyways it makes more sense to do it all in a single go instead of retooling it now and then rebuilding it later anyways.  Unless there is a major problem presented by not doing this now I saw we punt.

brentpicasso, 2016-07-10:
I agree, we can potentially address along side #172

brentpicasso, 2017-04-19:
Done in 2.11.0

closed 2017-04-19 by brentpicasso