autosportlabs/RaceCapture-Pro_firmware #145: Add ability to support scaled value interpolation via the use of functions.  Could help replace the mapping system we have for analog gauge values.

ryandoherty, 2015-10-26:


stieg, 2016-01-04:
Rather than a function we should provide a multiplier and an addition field provided to the user so they can specify conversion parameters to a real value.  Also should enforce min and max values to prevent wild values from occurring.

brentpicasso, 2016-01-04:
This is already in the firmware, just not exposed through the RaceCapture App UI. The multiplier field is exposed, but not the adder.

Re-open if any other concerns.

stieg, 2016-01-04:
So this is all plumbed in then to the JSON API @brentpicasso ?  If so then excellent.  Else lets keep it open until it is usable by the App.

brentpicasso, 2016-01-04:
Yes, it is implemented as the offset field in the JSON API.

stieg, 2016-01-04:
@brentpicasso Can you help me understand the implementation since I don't see any docs on it.  Here is what I see that may be relevant in the code:

```C
static const jsmntok_t * setAnalogExtendedField(const jsmntok_t *valueTok, const char *name, const char *value, void *cfg)
{
    ADCConfig *adcCfg = (ADCConfig *)cfg;
    if (NAME_EQU("scalMod", name))
        adcCfg->scalingMode = filterAnalogScalingMode(modp_atoi(value));
    else if (NAME_EQU("scaling", name))
        adcCfg->linearScaling = modp_atof(value);
    else if (NAME_EQU("offset", name))
        adcCfg->linearOffset = modp_atof(value);
    else if (NAME_EQU("alpha", name))
        adcCfg->filterAlpha = modp_atof(value);
    else if (NAME_EQU("cal", name))
        adcCfg->calibration = modp_atof(value);
    else if (NAME_EQU("map", name)) {
        if (valueTok->type == JSMN_OBJECT) {
            valueTok++;
            valueTok = setScalingRow(adcCfg, valueTok);
            valueTok = setScalingRow(adcCfg, valueTok);
            valueTok--;
        }
    }
    return valueTok + 1;
}
```

and

```C
float get_analog_sample(int channelId)
{
    LoggerConfig * loggerConfig = getWorkingLoggerConfig();
    ADCConfig *ac = &(loggerConfig->ADCConfigs[channelId]);
    float value = ADC_read(channelId);
    float analogValue = 0;
    switch(ac->scalingMode) {
    case SCALING_MODE_RAW:
        analogValue = value;
        break;
    case SCALING_MODE_LINEAR:
        analogValue = (ac->linearScaling * (float)value);
        break;
    case SCALING_MODE_MAP:
        analogValue = get_mapped_value((float)value,&(ac->scalingMap));
        break;
    default:
        analogValue = -1;
        break;
    }
    return analogValue;
}
```

In there I can see where we use the "scaling" value to apply the linear scaling.  However I don't see any use of the "offset" value.  From what I see this isn't completely implemented yet.

stieg, 2016-06-05:
Merged

closed 2016-06-05 by stieg