autosportlabs/RaceCapture-Pro_firmware #161: Add RC URL to headers in console.

ryandoherty, 2015-10-26:


meikeric, 2016-05-03:
I'll start with this one... What exact URL do you want it to point to?

brentpicasso, 2016-05-03:
Hi, thanks. I'm not even sure what this issue is for we might close it out. Anyone?

stieg, 2016-05-03:
What @ryandoherty wanted here was to have our URL for our site appear in the interactive serial console.  So when you connect via serial and hit enter twice you will see the welcome screen.  He wants the URL in there.

meikeric, 2016-05-03:
So... www.autosportlabs.com? www.podium.live?

brentpicasso, 2016-05-03:
Oh, adding something to the welcome message. So, it would be http://www.autosportlabs.com then.

ryandoherty, 2016-05-03:
This ticket was created by my account when we moved from Trello to GH, but I didn't create the original task.

closed 2018-05-17 by brentpicasso