autosportlabs/RaceCapture-Pro_firmware #165: Wifi module support

ryandoherty, 2015-10-26:
support the ESP8266 module on the wireless or Aux port. possibly have an auto-detect mechanism.

zkdzegede, 2015-12-24:
I'm working on spoofing the BT connection using this chip but having trouble reading the UART commands.  We're you able to get that part up and working? 

I can send and receive from the esp tx/tx to itself and receive messages, but when I try the same with the rcp it cannot even read its own messages. I'm not sure what's going on... Any idea?

brentpicasso, 2015-12-26:
Hi, also posted on the forums.

For connecting things to the RJ11, make sure you're using RS232 signaling. regular UART signaling won't work (and may damage components that expect just UART / TTL signaling ). 

A good test would be to USB to serial adapter to the serial port and test it with a terminal program.

stieg, 2016-06-05:
This is currently working (albeit not complete).  Going to close this

closed 2016-06-05 by stieg