autosportlabs/RaceCapture-Pro_firmware #169: Speed based auto-start/stop logging

ryandoherty, 2015-10-26:


brentpicasso, 2016-02-10:
Also - 

"is engine running" auto logging option.  Could be based on battery voltage, or RPM channel

stieg, 2016-05-16:
Lets try and sneak this in as part of 2.10

brentpicasso, 2016-05-16:
agreed. it's time.

stieg, 2016-07-07:
So for this revision of the feature I am going to do speed based auto start/stop only.  RPM adds complications because I don't know which channel to monitor or which channel is the correct RPM channel to monitor (since there is a chance there could be more than one).

The main question I have now are what our default values should be.  Should this be enabled by default since its a new feature and users are not used to having it?  I can see both sides of the fence here really.  On one hand if they don't have an SD card installed we will try to log to something that doesn't exists and that will turn on the red light and may create user issues.  On the other hand we want things to "just work".  I'm typically of the camp of "make the user enable the feature" so they explicitly know they did it.  Tends to create a bit less confusion that way.  Thoughts?

ryandoherty, 2016-07-07:
+1 for just speed based for now. Simple and effective.

Defaulting to on for current customers could cause problems, but defaulting to off will make our new customers' experience worse. If we didn't have LuaScript, I'd be 100% behind enabling it by default. 

I'm thinking we have the default of it being on and log when speed is > 5MPH. We'd need some very strong messaging for users. 

Having a red light turn on if there's no SD card is fine, by default users will want to log data so RCP should tell people when it can't. If they don't want SD card logging, they can turn it off.

brentpicasso, 2016-07-07:
I agree making this simple for now, advise on designing the configuration API with extensibility in mind.

To trigger on other sensors we could maintain a list of other sensor inputs to trigger on and make that configurable by sensor index (0-8 for analog) (0-2 for timer) (0-2 for GPIO) etc. Start and stop thresholds would likely be the same.

ryandoherty, 2016-07-07:
Also, we need a spec for the API message to configure this so we can get it into 1.7.0 (we may need to delay 2.10.0 to allow us to release a new app)

brentpicasso, 2016-07-07:
Also, agree on having it on by default as well.

jeffmcaffer, 2016-07-07:
Random user point of view...  On by default is fine for me.  I get to delete that part of my Lua script.  For trigger speed, I tend to go for something higher (e.g., 20mph).  Less and it triggers just driving around the paddock and to/from the hot pits making lots of little log files to delete.  If I'm heading out on the track 20 is plenty early.  Plus >5mph && > 1s likely is >20mph :-)  For launch/autocross scenarios that may be different.

stieg, 2016-07-08:
Thanks for the feedback.  Currently it is off by default.  I will change this to on by default.  As for defaults here is what I currently have:

```c
#define DEFAULT_START_SPEED_MPH 25
#define DEFAULT_START_TIME_SEC  5
#define DEFAULT_STOP_SPEED_MPH  3
#define DEFAULT_STOP_TIME_SEC   30
```

This in code translates to the following:

* If you maintain speed >= 25MPH for more than 5 seconds, logging will turn on.
* If you maintain speed <= 3MPH for more than 30 seconds, logging will turn off.

That sounds pretty reasonable to me but am open to suggestions accompanied by reasoning.

stieg, 2016-07-08:
Changed stop speed and time to the following:

```c
#define DEFAULT_STOP_SPEED_MPH  15
#define DEFAULT_STOP_TIME_SEC   10
```

stieg, 2016-07-11:
Merged

closed 2016-07-11 by stieg