autosportlabs/RaceCapture-Pro_firmware #175: text based alert statuses

ryandoherty, 2015-10-26:
add a text field for the status API that can be set from Lua

id = addAlert(severity, message)
removeAlert(id)
clearAlerts()

Up to 100 characters?

stieg, 2016-01-04:
I can imagine this getting really annoying really quickly.  Is this a real use case we need to support?

brentpicasso, 2016-01-04:
This is a place holder to allow RCP to send some kind of up stream alert
notification. But I don't like it b/c RCP shouldn't be aware of any
particular display, adhering to separation of concerns. Still, if a coded
alert condition could be broadcasted that could be useful.

stieg, 2016-01-04:
IMHO this feature isn't best on RCP itself.  Rather this should be some form of RCL feature that is easily configured using a web UI.  The cost to benefit ratio of implementing this feature in firmware seems too low to be worth it.  Would you agree?  If not, please re-open.

closed 2016-01-04 by stieg