autosportlabs/RaceCapture-Pro_firmware #178: Change distance to be calculated by integrating speed over time.

ryandoherty, 2015-10-26:
Customer feedback: http://www.autosportlabs.org/viewtopic.php?p=22581#22581

stieg, 2016-01-04:
Will need to test if this is more accurate than what we currently have.

ryandoherty, 2016-01-04:
I think a user commented in the forums or via email a long time ago with data that showed it was (used RCP log to derive distance based on speed). We still need to test, but seems promising.

JJMcDonough, 2016-01-13:
I'm not sure what method can be used to verify accuracy, but spot checking one of my files I do see about a 3% difference in distance over 1 mile comparing the raw distance channel (GPS set to 10hz) to my speed integration distance calc. Like I said on the message board, when I was running too high of a GPS sampling rate, the raw distance was way off because the position data jumped around a bunch. Happy to provide file/feedback/testing.

brentpicasso, 2016-01-13:
Hi, great feedback.  Do you have a sense which one is more accurate? IOW, did you also compare this against a known distance?

JJMcDonough, 2016-01-16:
Did not compare against a known distance. I have DL1 data on the same course, but it's a bit tricky to identify the finish line from the data.

stieg, 2016-06-29:
merged with commit 18b423aea03d98f69c4a54f4d0aa341fe51688c7

closed 2016-06-29 by stieg