autosportlabs/RaceCapture-Pro_firmware #209: Bluetooth initialization problem

ryandoherty, 2015-10-26:
Bluetooh initializes too fast (race condition before BT module is ready)

stieg, 2015-11-16:
@ryandoherty Huh... Can you elaborate more on this?

brentpicasso, 2015-11-16:
I think I opened this, and corresponds to the delay before we try to talk
to the module. We need to wait some amount of time before we can
communicate with module.

The delay is not specified and the module does not output a startup message
AFAIK so we have to blindly wait.

Brent Picasso
Autosport Labs
Technology for Race and Street

stieg, 2015-11-16:
Ahh that is most certainly a part of this fix (it was already there, but added some pauses when we change the serial baud).

stieg, 2015-12-27:
Closing because merged

closed 2015-12-27 by stieg