autosportlabs/RaceCapture-Pro_firmware #210: need customer accessible downloads for nightly and released versions of software

ryandoherty, 2015-10-26:
RIght now we're hand-editing the wiki and hand uploading files to autosportlabs.net.  

We need a customer-browsesable set of pages that will show the things that can be downloaded.

idea: there are some apache modules that provide nice formatting for directory browsing. 

idea URL? http://autosportlabs.com/downloads

stieg, 2016-06-05:
Customers don't care enough right now.  Hence closing due to lack of interest.

closed 2016-06-05 by stieg

kosenko-max, 2016-06-06:
I wouldn't say they don't care. I don't know how many of them care about it
same as me...

stieg, 2016-06-06:
Fair enough... I should rephrase that as "Not enough interest has been shown to warrant the work".  If you want a nightly just ask.  If I get asked enough I imagine I might make the feature :D