autosportlabs/RaceCapture-Pro_firmware #344: Adjust BT name of RCP units to include last 4 digits of serial

stieg, 2015-11-12:
Needed because when multiple RCP units are around and you try to pair, it can be very difficult to tell which unit you are looking for.

brentpicasso, 2015-11-12:
We have APIs for setting the BT name, not yet exposed in the app.

If this feature is implemented, it's important to ensure this is only set
when restoring factory defaults.

stieg, 2015-11-12:
Agreed.  It would only apply if a factory reset happens.  That is how I will program the patch.

stieg, 2015-11-16:
Hrmm... interesting issue here.  The BT device name is set when the unit boots.  So if the unit boots and the BT name has been changed, it will be reflected during that boot.  That seems logical but maybe you may disagree @brentpicasso

stieg, 2015-12-27:
Closing because merged.

closed 2015-12-27 by stieg