autosportlabs/RaceCapture-Pro_firmware #348: Periodic glitch in Pitch channel

brentpicasso, 2015-12-01:
A periodic glitch value only observed in the Pitch channel is observed in this data set (attached)

See lines 375, 757.  Note how the value somewhat corresponds to the interval value.

[Rowan_RC_0.log.txt](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/48139/Rowan_RC_0.log.txt)

brentpicasso, 2015-12-01:
Also observed in the yaw channel on line 48056.

stieg, 2015-12-01:
What version of RCP firmware is this occurring on and what platform?  Is it constantly observable or just with specific settings in place?  Reproducible?  It looks like memory corruption, but is somewhat random.  Perhaps an index somewhere in a sample buffer is on the fritz.

brentpicasso, 2015-12-01:
Verifying, but quite certain it is 2.8.5 with this integration script running: https://autosportlabs.net/Link_G4

stieg, 2015-12-01:
And this is MK2 as well?

brentpicasso, 2015-12-01:
Correct, MK2

RowanH, 2015-12-02:
Just a note - I recorded 3 log files, and all 3 had the errors, unfortunately I overwrote the 2nd log file cleaning out the errors. I'll try without the CAN script running and leave it as default as a config as I can. It's inconsistent what channels get messed up first, sometimes AccelZ, sometimes RPM etc. 

First log file was minimal RCP config - I'd just applied the CAN script, no GPS 
Second and third log file I'd calibrated the accelerometer, and wired in the GPS. 

I'll see if I can test without the CAN integration.

closed 2018-05-17 by brentpicasso