autosportlabs/RaceCapture-Pro_firmware #351: Reduce GPS Baud Rate

stieg, 2015-12-14:
GPS baud rate on 2.8.5 and others is stupid high.  Something like 921600bps.  That is insane and unnecessary and is likely causing issues and many timeouts:

https://www.autosportlabs.org/posting.php?mode=reply&t=4615

I propose we reduce this to a lower rate to add stability.

brentpicasso, 2015-12-14:
The baud rate is set to the mfg recommended speed according to their datasheet. 921600 baud is not an unusual bad rate for transmitting between peripherals on the same board when components are within inches of each other (GPS module and processor).

If there are any errors or timeouts, it is very much likely not related to baud rate. Our initialization code requires a few attempts to synchronize to the stream when using DMA. The other possible issue is likely a defective GPS module, which has been observed in the past.

stieg, 2015-12-14:
@brentpicasso I see these disconnects on both my units somewhat frequently.  It may not be an unusual rate, but the symptoms I see appear to be ones seen when devices get out of sync with one another.  If its not related to baud, what would you attribute it to?  I suggest we try reducing the baud by 2 or 4 and see if it makes a difference in stability.  Defective GPS systems do happen, but the rate of which I see issues indicates something greater than that.

brentpicasso, 2016-01-17:
There's no evidence that reducing the baud rate improves anything, and we're using the baud rate currently recommended by the GPS module mfg.  Closing this. re-open if compelling reason to re-investigate it.

closed 2016-01-17 by brentpicasso