autosportlabs/RaceCapture-Pro_firmware #359: Reset distance channel to 0 when crossing the start

ryandoherty, 2016-01-02:
RaceCapture/Live is seeing lap data where a new lap starts, but the Distance channel's value starts at values much greater than 0 (0.2, 0.5, etc). This is most likely due to having separate start/finish points and RCP is 'counting' the distance between finish => start, making the Distance channel start at a high value.

This breaks a lot of functionality in other systems and also makes it impossible to compare lap data over distance.

ryandoherty, 2016-01-02:
Nominating this to be fixed/investigated for the next firmware release.

stieg, 2016-01-02:
Interesting.  I'll have a look.

ryandoherty, 2016-01-02:
Tough part is I saw this for telemetry data from a RC => phone => RCL connection, so no raw logs. Need to get the app recording data locally too.

stieg, 2016-01-03:
So far investigation has found minor issues in how start logic is handled.  A patch is in progress but not known if it will directly solve this issue.  We must be able to support us starting with a distance value > 0 (but not one that is .5 miles) since the implementation of the launch control feature requires us to start a lap with a non-zero distance (because launch control only knows you have started your lap _after_ it has detected a successful launch).

stieg, 2016-01-11:
Tested and passed as expected.

stieg, 2016-01-11:
ok to test

stieg, 2016-01-11:
test this please

stieg, 2016-01-18:
Need to port fix to 2.8.7

stieg, 2016-01-19:
PR: https://github.com/autosportlabs/RaceCapture-Pro_firmware/pull/393

closed 2016-01-19 by stieg