autosportlabs/RaceCapture-Pro_firmware #366: Conver API status codes into enum, update meanings.

stieg, 2016-01-07:
Currently the API status codes in firmware look like this:

```C
#define API_SUCCESS_NO_RETURN   2
#define API_SUCCESS                     1

#define API_ERROR_UNKNOWN_MSG   0
#define API_ERROR_PARAMETER     -1
#define API_ERROR_MALFORMED     -2
#define API_ERROR_SEVERE                -99
#define API_ERROR_UNSPECIFIED   -100
```

While functional this should be an enum type.  Also there doesn't seem to be any difference in the handling of `API_SUCCESS_NO_RETURN` vs `API_SUCCESS` so we should eliminate those differences.

stieg, 2016-01-07:
Also... why do we have a return type that is `API_SUCCESS_NO_RETURN`?  I see that returning this will suppress a return message.  I'm curious why we would want that?  @brentpicasso?

brentpicasso, 2016-01-08:
The API_SUCCESS_NO_RETURN  is used for API calls that returns data by design, like the get*** API calls. 

The set*** calls do not inherently return data, so the message that is returned is like an ack.

stieg, 2016-01-08:
Ahh... that makes sense now.  Rather than explicitly invoke a method to return a status method, you controlled that behavior using the return code.  Makes sense.

closed 2018-05-17 by brentpicasso