autosportlabs/RaceCapture-Pro_firmware #367: Fix firmware API so invalid JSON doesn't corrupt existing config

stieg, 2016-01-07:
In the firmware API calls if a method fails midway through its operation then the loggerConfig may end up in an invalid state since we directly modify the loggerConfig as we parse the message.

A saner approach would be to allocate a temporary buffer to place the new values into and then copy them over to the working config once we have successfully parsed and read the JSON.  This will prevent corruption if a message ends up being malformed.

brentpicasso, 2016-01-08:
We have a level of protection already with the way the API is processed. The API function will not be called if the JSON is malformed (it will immediately return an error).

If it is well-formed JSON, then the rest of the handling is fairly consistent throughout the system: the API function handling the message will look for data present in the correct location and pull values out, updating the config as needed. 

If a field is not present, no update occurs for that value. If extraneous data is present in the JSON, it is ignored.

Providing the API is consistent, I believe the current approach is safe to keep and does not require changing.

stieg, 2016-01-08:
> We have a level of protection already with the way the API is processed. The API function will not be called if the JSON is malformed (it will immediately return an error).

Yes.  Very true.

> Providing the API is consistent, I believe the current approach is safe to keep and does not require changing.

As the system currently stands, probably true.  But these are open APIs and there is some support for updating this config over other channels like the cellularModem (Though we don't currently use it).  But I agree this is a lower priority "nice to have" ticket.

brentpicasso, 2016-01-10:
Since we have an established pattern of required and optional parameters in
the JSON,

I think the effective action is to cover this in API documentation,
explaining which parameters are required and optional, and doing some
detailed testing as we go.  Any inconsistencies discovered during this
process could turn into separate issues as we discover them. The end result
would be a documented and consistently behaving API.

Thoughts?

stieg, 2016-01-10:
While I believe that documenting the API is important, I am not sure that this will solve this problem.  For example, lets say that someone provides a JSON blob that sets channel values 0 - 8.  This would return an `API_ERROR_PARAMETER` status because when the code would go to try and fetch the config for channel at index 8 (which doesn't exist), the code will fail (as it should).  The problem is that the values set before (on channels 0 - 7) has been updated, but since the call returns with `API_ERROR_PARAMETER`, the firmware ends up in an inconsistent state because we don't invoke the methods that tell the code that the config has changed.  This means that the changes won't be reflected in the behavior of the firmware, even though it has been changed.  And while this is not _so_ bad, its not good.  Ideally the firmware would not change any of its values if an invalid JSON blob is passed in and processed.

closed 2018-05-17 by brentpicasso