autosportlabs/RaceCapture-Pro_firmware #374: Enable printf logging to SD card

stieg, 2016-01-11:
Currently our logging only goes to a circular buffer on the RCP device.  We need a way to have the log get written to the SD card for easier analysis.  Need to consider the following:

* Need a switch in the API to enable/disable this feature
* Logs need to go into their own directory on the SD card FS
* Need to enable support for many writers to the SD card at the same time.

chriva, 2016-02-23:
I'd recommend FatFS. It can handle several requests at once but I don't think it's capable of doing so with several files open at the same time. Ie Can handle many request but only to one file at a time.

I have some code for stm32f1 and f3 that takes care of the hardware part on those microcontrollers if you guys have use for it?

stieg, 2016-02-23:
@chriva currently we use FatFS, but the way we use it is designed around logging data instead of logging data _and_ logging log files.  Effectively we need move away from this legacy model and start supporting multi-file writing internally.  Figuring how to recognize and handle SDcard disconnects and the re-insertions is paramount if possible.  If you have code that solves that problem I would be interested to know where it lies.

stieg, 2016-04-25:
Depends on #511

closed 2018-05-17 by brentpicasso