autosportlabs/RaceCapture-Pro_firmware #379: Fix BT password regression introduced with new BT features.

stieg, 2016-01-13:
Need to fix the regression @brentpicasso found.  Outlined in the following chat:

```
​[22:09] 
brent Ok, I tried again and saw something really weird.

I had already paired the RCP Bt module in android.  (specified pwd, everything)..  I relaunched the app and when I ran the app android challenged me to re-pair and specify password ​*again*​.  After that, I was able to connect again no problem
​[22:10] 
App connected and read data from RCP
​[22:10] 
That's very odd
​[22:11] 
I wonder if it has something to do with the 0000 pin.. I thought I remembered there was something special about that number.
​[22:12] 
Yes, I power cycled the BT module and it did it again
​[22:12] 
after re-launching the app
​[22:16] 
Correction: I power cycled RCP which caused it to send the BT module new configuration commands
​[22:16] 
​*that*​ causes android to issue the password challenge again
​[22:16] 
I'll take a video.
​[22:32] 
ryan That is really craY
​[22:40] 
brent I did a quick test of disabling the setting of password
​[22:40] 
apparently setting the password again (even if it's the same value) causes the problem
​[22:42] 
stieg Oh... Interesting
​[22:42] 
Ok... That should be easy to fix
```
Demo'd here: https://www.youtube.com/watch?v=cu9ZFWHxg1o

stieg, 2016-01-13:
Wonderful.  It would seem that there is no way to read the current PIN using AT commands from the device.  Here is the full documentation: https://www.olimex.com/Products/Components/RF/BLUETOOTH-SERIAL-HC-06/resources/hc06.pdf

So we either need to simply leave the default password or need to come up with a creative way to know that the password changed and thus needs to be updated.

brentpicasso, 2016-01-13:
Yeah, it's crap. :(

Before, the one way I knew the module was not provisioned was to check if it's at 9600 baud. If it was at 115200, I assumed it was already provisioned.

Maybe setting the Name / PIN could be an explicit action.

stieg, 2016-01-14:
#383 should solve those issues.  Similar to what you described @brentpicasso

stieg, 2016-01-15:
Fixed and verified.

closed 2016-01-15 by stieg