autosportlabs/RaceCapture-Pro_firmware #403: If Lua script is too busy, cannot update firmware

brentpicasso, 2016-01-25:
If Lua script is too busy (e.g. never returning from the onTick() function) the script cannot be updated due to not being able to get a lock on the Lua runtime. 

Work-arounds: always provide a way for the onTick() to exit after a reasonable amount of time.

stieg, 2016-01-27:
We can create a simple Lua_Watchdog task for this.  Using a semaphore if the LUA task takes too long then we can kill the execution of the task and re-start it (or something along those lines).

kosenko-max, 2016-03-12:
Is it possible to keep ability for script to never go out of onTick()? It's the only way how I can consistently pull CAN data without gaps. All other approaches put huge timing gaps in the log.

Maybe kill Lua once script update is required? Instead of killing it by watchdog?

stieg, 2016-03-12:
> Is it possible to keep ability for script to never go out of onTick()? It's the only way how I can consistently pull CAN data without gaps. All other approaches put huge timing gaps in the log.

Well starting with 2.9.0 you will have support for setting the onTick invocation rate to be as high as 1000Hz.  So really you will never need to worry about leaving `onTick` starting with 2.9.0.

> Maybe kill Lua once script update is required? Instead of killing it by watchdog?

Yeah that is the plan of attack in general.  Right now the Lua task isn't well suited for destruction and re-init so some work would need to happen to make it better suited for this use case.

stieg, 2016-04-11:
In 2.9.0.

closed 2016-04-11 by stieg