autosportlabs/RaceCapture-Pro_firmware #404: Lua txCAN timeout parameter has no effect

brentpicasso, 2016-01-26:
The timeout message is available in the lua function, but has no effect down in the STM32 HAL. 

need a way to enable a timeout, or remove the timeout parameter.

Also - in the HAL layer the CAN_device_tx_msg always returns 1; should return 1 or 0 indicating success.

stieg, 2016-01-27:
Seems that the timeout portion was skipped in the hal implementation.  I see methods for CAN message status and cancellation in the stm32f4xx_can.h file:

```C
/* CAN Frames Transmission functions ******************************************/
uint8_t CAN_Transmit(CAN_TypeDef* CANx, CanTxMsg* TxMessage);
uint8_t CAN_TransmitStatus(CAN_TypeDef* CANx, uint8_t TransmitMailbox);
void CAN_CancelTransmit(CAN_TypeDef* CANx, uint8_t Mailbox);
```

So probably best to implement the timeout here in HAL.

stieg, 2016-01-27:
This is a 2.9.0 feature since it changes behavior (albeit for the better).

brentpicasso, 2016-01-27:
Yeah. the Rx method conveniently uses the FreeRTOS queue for timeout.

Since only Lua uses the txCAN function (so far) - we can implement a cheap
sleep() delay loop up until the timeout if the STM32 CAN transmit function
fails to actually send the message. Wouldn't be terrible.

-Brent

stieg, 2016-01-31:
Patch tested and merged.

closed 2016-01-31 by stieg