autosportlabs/RaceCapture-Pro_firmware #419: Investigate Way to Have Driver Alerts Go Through RCP

stieg, 2016-02-09:
There is interest in the ability for people to be able to send alerts to the tablet in the vehicle during a race.  Alerts could be anything from flag indicators to a few words (like "PIT").  This could be achieved by creating a new message type in RCP that would get internally echo'd to all telemetry streams (BT / WiFi).  Then the apps would just have to know how to handle them.

brentpicasso, 2016-02-09:
I was thinking of a broadcast mechanism too for alerts.  Would also be beneficial to find a way for driver to ack message, so the pits knew they saw it. 

Could also incorporate a simple Yes / No option as part of confirmation.

stieg, 2016-02-09:
Definitely possible.

closed 2018-05-17 by brentpicasso