autosportlabs/RaceCapture-Pro_firmware #423: Add unlimited Tick Rate option

kosenko-max, 2016-02-11:
Currently I can pull around 60 samples a second for each of 45 CAN channels (they are really packed in a messages - http://pastebin.com/02i5Dp6L). But I can't update it that fast in MK2. Is there any way to actually remove a limit of ticks, but instead do limit of time spent in lua scripts based on demand? Real tick rate in such case would be great to see as a channel so that script optimizations could be done.

Thanks.

stieg, 2016-02-11:
Check out ticket #172 as that will likely be of interest to you.  That will enable firmware based CAN reading, which will be must faster than LUA.

The rate limit of LUA is mostly due to historical reasons (MK1).  LUA can be run in a non-ending loop if you choose (just never return from the onTick method), but I'm not totally sure what will happen if you do that.  It shouldn't starve any of the other tasks as LUA is the lowest priority task.  I suggest you try using an infinite loop in LUA and use of the LUA sleep method to iterate over all CAN bus messages, then sleep for 5ms, then do the loop again.  Things _should_ work, but since its not a supported (and tested ) use case I can't guarantee it.

Interested to know what you find.

kosenko-max, 2016-02-11:
I will try to do that. Thank you. I just worry about recent ticket that Lua can't unload and update code in such case. But in theory that would be best.

In most cases that would not be endless math in the ticks, so it will be sleeping on a calls to CAN/serial or other ports anyway.

stieg, 2016-02-11:
Ahh... well that is easy to work around.  Set ticks to 10, then loop like 20 times with a 5ms break in between, then break that loop.  That means each iteration should take about 100ms, which means the call should happen again shortly.

stieg, 2016-02-11:
I should clarify.. the code is set to call the `onTick` method every X ms.  It determines the next call time before it calls the `onTick` method so that variations in the Lua run time don't affect the periodic callback timing.  So if you use the above approach, it should call back at 100ms intervals, and each loop should take about 100ms to run.  perfect.

kosenko-max, 2016-02-11:
I think I can try to loop just above tick timer without sleep (I'm sure CAN would sleep for me more than enough) by checking getUptime(). From the code it looks like it will call onTick right away (yes, you just clarified that above right now).

stieg, 2016-02-11:
> In most cases that would not be endless math in the ticks, so it will be sleeping on a calls to CAN/serial or other ports anyway

Yes this is true.  So long as you aren't using both LUA and the OBD-II functionality this will work.

> I just worry about recent ticket that Lua can't unload and update code in such case.

Very good point.  Forgot about that.  Well the above will work around that issue :).

kosenko-max, 2016-02-11:
Thank you - I will try that hack tomorrow and let you know how it works.

kosenko-max, 2016-02-13:
Tried. It works, but I see that now every second update process stucks for like 0.1 second. Before it was only on USB connection, now on Bluetooth as well. Setting tick rate like 0.1 almost hangs a system. Shorter ticks decreasing refresh rate.

brentpicasso, 2016-02-13:
Hi Max, glad to see it's working.

Just to clarify, the parameter to setTickRate() is specified in Hz.  The
default is 1 Hz.

Brent Picasso
Autosport Labs
Technology for Race and Street

kosenko-max, 2016-02-13:
Found a reason for freezes - it was running out of memory after some time and hard restarting Lua.

stieg, 2016-02-21:
We can't necessarily do an unlimited rate, but effectively the same would be the maximum supported log rate, which is 1000Hz.  I assume this would suffice.  With our priority levels setup correctly (which they should be at this point), we can safely do this now since Lua runs at the lowest priority (and hence can not starve other tasks from running).

stieg, 2016-02-22:
PR #449 addresses this.

stieg, 2016-02-22:
Merged

closed 2016-02-22 by stieg