autosportlabs/RaceCapture-Pro_firmware #424: addChannel("Something", 30) hangs Lua silently

kosenko-max, 2016-02-11:
addChannel("Something", 30) hangs Lua silently - it took a lot of time to find out what's going on since there were no messages and system was not even accepting code updates and constantly restarting.

stieg, 2016-02-11:
Well that is annoying.  Will investigate.

brentpicasso, 2016-02-11:
I think it relates to the 30 of the sample rate you specified. Does it
happen with one of the supported sample rates?

https://autosportlabs.net/RaceCapturePro_Lua_Scripting#addChannel.28_name.2C_sampleRate.2C_.5Bprecision.5D.2C_.5Bmin.5D.2C_.5Bmax.5D.2C_.5Bunits.5D_.29

It looks like it returns a disabled sample rate, which may be causing the
hang downstream. Looks like we need to better validate the specified sample
rate and fail the addChannel() function as necessary.

kosenko-max, 2016-02-11:
Do you actually need any update rate specified on a virtual channel? I struggle to understand any meaning of it at all...

stieg, 2016-02-11:
Yes you do.  The logging mechanism takes samples at X hz.  When you update a rate on a virtual channel, that will update the sample value that will be read by the logger.  Since these are independent processes you need to tell the logging mechanism how often you want the sample updated in the telemetry stream.  It assumes that you will be updating it at that rate.  Otherwise it just sends the previous value that it got.

kosenko-max, 2016-02-11:
I see. Once you log anything at 1000Hz - is it actually going to make any difference that other channels  are not that fast? You still need to try to make 1000 lines a second, so you save only on reading values from the table of channels. But channels anyway updated at their own pace.

It looks to me that having a single log rate for everything and adjust rest of the stuff to that rate as much as possible would be way more clear approach for some future versions. It makes sense to give some priorities for CAN requests since they fight for same bandwidth. Well, I still didn't checked which processor you use, so maybe that could be a limit then?

stieg, 2016-02-11:
> I see. Once you log anything at 1000Hz - is it actually going to make any difference that other channels are not that fast?

Only disk space and bandwidth become your bottlenecks when you log at those rates.  And if its live telemetry your back end also can become an issue (think of scale for both writes and reads among lots of users.  It adds up quick).

> It looks to me that having a single log rate for everything and adjust rest of the stuff to that rate as much as possible would be way more clear approach for some future versions.

Again... bandwidth limitations are a key reason things aren't this simple.  If you log at 1KHz and each sample is 512B in text, you are writing 1000 * 512 * 60 = 30720000 = 30MB every minute.  That is a lot of data to chew through.

You are not wrong; it is most certainly possible to simplify by having a single rate of recording.  But the simpler design does not outweigh the consequences described above.  So for now we have to support multiple streaming rates.

> It makes sense to give some priorities for CAN requests since they fight for same bandwidth. Well, I still didn't checked which processor you use, so maybe that could be a limit then?

All inputs have the same processor priority level in our firmware.  CAN messages are delivered using an interrupt mechanism so there really is no risk in dropping messages unless the CAN bus just gets absolutely hosed with data, but then there are other issues to contend with.

Modern processor (MK2 +) can handle the load.  Wouldn't worry there.

kosenko-max, 2016-02-13:
Thank you. I think space is not really a big issue. Even at 512B/line at 1000Hz you can have above 18 hours of non stop logs on a $9 price of 32GB microSD. But with some optimizations you can store it way more compact with a simple export during download from Podium that converts logs to CSV and makes a zip file (or even better - automatic export to the google drive files).

1. Define float vs integer type for channels and store it in binary form up to 4 bytes a sample
2. Keep values in variable size float/integer (there are libs for that) and store differential value from previous row instead of absolute. That would make average data below 1 byte a sample.
3. Skip repeating column values - bitmask at the beginning of a row can give a hint of what is omitted/included. Row starting bit shows if list of changed values is the same as on a previous row. That would cut at least 90% of data at 1000Hz.

150 channels * 8 bits  * 10%  * 1000 samples a second = 120 kbit/sec stream.
Small enough to send over BT and slow internet.

For data storage it's 51.5MB/hour. Or 636 hours on a cheap MicroSD card.
With auto logging start and 4 hours/day average car time it's almost **half a year full speed logging** before any old logs should be automatically cleaned up.

And that is even without any noise filtering or dictionary based compression algorithms. So, technically, almost no performance hit at all. And I'm sure that my estimations are conservative. 

I can later do a tool that converts CSV logs to such format to show you size difference if you would really consider to do such an upgrade.

kosenko-max, 2016-02-13:
I will do log packing tool that works without forward looking logic in a limited memory usage to run over real existing log files. If you will find that it's worth doing - you can use same code to change how you transmit and store log data. I have better ideas than the one I described above, but need to verify and get real world data. In theory it would work especially effective on a very high speed logs so that no update speed limit would be necessary for any purpose.

stieg, 2016-02-25:
merged

closed 2016-02-25 by stieg