autosportlabs/RaceCapture-Pro_firmware #427: Lua Memory Size Limit on MK2 is too small

kosenko-max, 2016-02-13:
I've minified a code which is 25% of what I expect to have and it doesn't fit in 50kB anymore. What is the memory available at MK2 so that limit have to be so low? http://pastebin.com/02i5Dp6L

kosenko-max, 2016-02-13:
I'm not going to fit in both of those limits. Minified Script size is 9kb, but it will be 4 times larger.
And memory is even harder limit. 

#define SCRIPT_MEMORY_LENGTH	(1024 * 16)
#define LUA_MEM_MAX (1024 * 50)

Should I abandon further development or it was set too conservative and you have plenty of onboard memory?

kosenko-max, 2016-02-14:
You probably know those, but just in case:
http://www.eluaproject.net/doc/v0.9/en_elua_egc.html
http://www.eluaproject.net/doc/v0.9/en_arch_ltr.html

kosenko-max, 2016-02-14:
Also, probably could use precompiler in the app http://www.lua.org/manual/5.1/luac.html to avoid OOM during loading stage.

stieg, 2016-02-18:
As discussed #414, the memory ceiling for Lua is 50K because total available system RAM is 192K.  I will look into ways to cut down on the memory usage in Lua, but I will track that work in #414 since that issue has more in-depth conversation.

closed 2016-02-18 by stieg