autosportlabs/RaceCapture-Pro_firmware #431: Be able to filter out errant GPS data if possible

stieg, 2016-02-16:
Sometimes the GPS unit sends errant data (like speed going from 80 -> 125 -> 80 across 3 samples).  Ideally we would have a way to detect this and compensate for it.  Currently we take the data reported by the GPS at face value, and thus when issues like this hit we get issues (like distance being way off etc).

@ryandoherty Will you provide a link to some sample data to help with debugging?  Cheesy had these issues at Sonoma, so perhaps a couple of laps would be good to help develop a solution.  TIA.

kosenko-max, 2016-05-02:
Sorry, forgot to check for duplicates #545

stieg, 2016-05-02:
Tis good.  I like the idea of getting a Kalman filter implemented and tested first, then we can solve this issue with a Kalman (pending that it works).