autosportlabs/RaceCapture-Pro_firmware #441: Vagrant development environment setup problem

kosenko-max, 2016-02-19:
During "vagrant up":

> ==> default: Mounting shared folders...
>     default: /vagrant => /Users/KosenkoMax/Documents/workspace/RaceCapture-Pro_firmware
> ==> default: Running provisioner: shell...
>     default: Running: inline script
> ==> default: stdin: is not a tty
> ==> default: /tmp/vagrant-shell: line 1: /vagrant/vagrant_setup.sh: No such file or directory
> The SSH command responded with a non-zero exit status. Vagrant assumes that this means the command failed.

vagrant_setup.sh is in the bin/ folder instead.

After manual run still doesn't work - shared folder becomes empty from Guest OS.

stieg, 2016-02-19:
Fun.  Haven't used it in a bit since I moved back to Linux land.  Will fire it up and check it out.

kosenko-max, 2016-02-19:
Also fails at hard linking afterwards.