autosportlabs/RaceCapture-Pro_firmware #442: Auto-calculate the closest CAN baud rate prescalers based on the the specified baud rate

brentpicasso, 2016-02-20:
An algorithm can be used to select the closest prescaler for nearly any specified CAN baud rate. The system can be more versatile if an approach like this is used. 

Some helpful info:
https://my.st.com/public/STe2ecommunities/mcu/Lists/STM32Discovery/Flat.aspx?RootFolder=https%3a%2f%2fmy%2est%2ecom%2fpublic%2fSTe2ecommunities%2fmcu%2fLists%2fSTM32Discovery%2fCAN%20Baud%20rate&FolderCTID=0x01200200770978C69A1141439FE559EB459D75800084C20D8867EAD444A5987D47BE638E0F&currentviews=6266

chriva, 2016-02-21:
This fella works very well en pretty much anything. Originally made it for stm32 but have since used it on mcp2515, tms570, elm327 and lawicel canusb with minor modifications.
/////////////////////////////////////////

uint16_t prescaler;
uint8_t  tseg1;
uint8_t  tseg2;
uint8_t  sync;

//These are parameters for 'checkcalc'//
#define DES_SAMP_PNT   75 //Desired sampling point.
#define MAX_SAMP_DEVI    10 //Allowable sampling point deviation.
#define MAX_BTR_DEVI    0.4 //Allowable percentage of bitrate deviation
#define PCLK1            42000000

uint8_t checkcalc(uint64_t desiredrate){

   //Check if current settings leads to desired bitrate.
   if       (((PCLK1)/prescaler) / (sync+tseg1+tseg2+3) > (desiredrate+((desiredrate/100)*MAX_BTR_DEVI))){return 0;}
   else if (((PCLK1)/prescaler) / (sync+tseg1+tseg2+3) < (desiredrate-((desiredrate/100)*MAX_BTR_DEVI))){return 0;}

   //If we have a bitrate match, check if those settings leads to a reasonable sampling point.
   else if (((100*(sync+tseg1+2)) / (sync+tseg1+tseg2+3)) > DES_SAMP_PNT+MAX_SAMP_DEVI){return 0;}
   else if (((100*(sync+tseg1+2)) / (sync+tseg1+tseg2+3)) < DES_SAMP_PNT-MAX_SAMP_DEVI){return 0;}

   //We've passed!
   else{
      return 1;
   }
}

////////////////////////////////////////////////////////////////////////////////////////////
//This thing will loop through all possible permutations until it finds suitable settings//
//(Or loop forever if it's unable to)/////////////////////////////////////////////////////
void STM32_Calc_BTR(uint64_t btr){

   ///< Reset stuff just in case this is not the first time this function is entered
   prescaler=0;
   tseg1=0;
   tseg2=0;
   sync=0;

   if (btr>10){ //Just to make sure user asks for a sensible bitrate.
      do{
         //sync++;
         /*if (sync>=1)*/{
            sync=0;
            tseg2++;

            if (tseg2>=7){
               tseg2=0;
               tseg1++;

               if(tseg1>=15){
                  tseg1=0;
                  prescaler++;

                  if(prescaler>=1024){
                     prescaler=0;
                     (leave function and fail)
                     (or increase allowable bitrate deviation)
      }}}}}while(!checkcalc(btr));
   }
   
   (Put code here to re-init the controller)
}

jordan81, 2016-05-27:
Any progress on implementing this code? I'd love to be able to upgrade the firmware to be able to use the newer features and wifi support. It is something that will be very important for selling the new RC when it comes out.

stieg, 2016-05-27:
> Any progress on implementing this code?

Nothing as of yet.  When bugs are in the process of being fixed they get assigned to the person creating the fix.  We are quite busy with other deliverables so keep an eye on this ticket.

jordan81, 2016-05-31:
Thanks. I understand, just checking in.

brentpicasso, 2018-06-14:
After discussing offline with @jordan81, the quickest, least regression solution would be to add an entry for 618K baud, with the following settings for all devices using the STM32F4 processors. 

STM32F3 (RaceCapture/Track MK1) will not be supported as the peripheral does not allow us to get close enough to the target baud rate. 

Prescaler: 17
TSEG1: 2
TSEG2: 1

brentpicasso, 2018-06-14:
Will add to the 2.13.5 firmware release.

brentpicasso, 2018-06-15:
Bumping this to 2.14 firmware, too much regression risk and we need to get the firmware out to address critical bug fixes.

brentpicasso, 2018-06-15:
@jordan81  - there are 4 CAN setup parameters, can you test and verify the 4 parameters so we can get it implemented? 

Thanks,
-Brent
//For 168MHz clock
/*       BS1 BS2 SJW Pre
 1M:      12  8   1   2
 500k:    8   5   1   6
 250k:    8   5   1   12
 125k:    12  8   1   16
 100k:    12  8   1   20 */

jordan81, 2018-06-26:
I think you covered this via email, but for posterity  it is:
618k: 2 1 1 17 

Thank you!