autosportlabs/RaceCapture-Pro_firmware #455: Failing to receive 29 bit CAN IDs on LUA?

stieg, 2016-02-27:
A user is reporting that 29 bit addressed messages are unable to be received in Lua land.  Here is the full conversation: 

https://www.facebook.com/AutosportLabs/messages/?threadid=1563768273&mercurythreadid=user%3A1563768273&timestamp=1456597001419

and script:

```lua
--This example configured for E46 CAN

--how frequently we poll for CAN messages
tickRate = 30
--the CAN baud rate
CAN_baud = 500000
--CAN channel to listen on. 0=first CAN channel, 1=second
CAN_chan = 1
--1 for Big Endian (MSB) mode; 0 for Little Endian mode (LSB)
be_mode = 0

--add your virtual channels here
tpsId = addChannel("TCK1_ext", 10, 0, -50, 1200, "C")

--customize here for CAN channel mapping
--format is: [CAN Id] = function(data) map_chan(<channel id>, data, <CAN offset>, <CAN length>, <multiplier>, <adder>)
CAN_map = {
--my id is 29 bit hex 10DB3500
[282801408] = function(data) map_chan(tpsId, 1, 1, 2 , 0, 0,tck1) end,

}

function onTick()
    processCAN(CAN_chan)
 
end

--===========do not edit below===========
function processCAN(chan)
    repeat
        local id, e, data = rxCAN(chan)
        if id ~= nil then
            local map = CAN_map[id]
            if map ~= nil then
                map(data)         
            end
        end
    until id == nil
end

--Map CAN channel, little endian format
function map_chan_le(cid, data, offset, len, mult, add)
    offset = offset + 1
    local value = 0
    local shift = 1
    while len > 0 do
        value = value + (data[offset] * shift)
        shift = shift * 256
        offset = offset + 1
        len = len - 1
    end
    setChannel(cid, (value * mult) + add)
end

--Map CAN channel, big endian format
function map_chan_be(cid, data, offset, len, mult, add)
    offset = offset + 1
    local value = 0
    while len > 0 do
        value = (value * 256) + data[offset]
        offset = offset + 1
        len = len - 1
    end
    setChannel(cid, (value * mult) + add)
end

map_chan = (be_mode == 1) and map_chan_be or map_chan_le
initCAN(CAN_chan, CAN_baud)
setTickRate(tickRate)
```

Need to validate that 29 bit CAN Rx is fully operational.

Fabrizio82, 2016-03-01:
Some update, i create a new script file:`-------------------------------------
--------------SETTING----------------
-------------------------------------
--how frequently we poll for CAN messages
tickRate = 30
--the CAN baud rate
CAN_baud = 500000
--CAN channel to listen on. 0=first CAN channel, 1=second
CAN_0 = 0
CAN_1 = 1
-------------------------------------
--------------INIT----------------
-------------------------------------
--Init both can bus
initCAN(CAN_0, CAN_baud)
initCAN(CAN_1, CAN_baud)
--Set tickRate
setTickRate(tickRate)
-------------------------------------
-----------Can Bus LOG---------------
-------------------------------------

function outputCAN()
	repeat 
		id, ext, data = rxCAN(0, 100)
		if id ~= nil then
		print('Can_0: ')
			print(id ..', ')
			for i = 1,#data do
				print(data[i] ..', ')
			end
			println('')
		end
		id, ext, data = rxCAN(1, 100)
		if id ~= nil then
		print('Can_1:')
			print(id ..', ')
			for i = 1,#data do
				print(data[i] ..', ')
			end
			println('')
		end
		
	until id == nil
end

function onTick()
	outputCAN()
end`

and i send over both can , one 11 and one 29 bit message:

[image]

and this is what i can pull from the log:

Can_0: 100.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 
Can_1:200.0, 2.0, 2.0, 2.0, 2.0, 2.0, 2.0, 2.0, 2.0, 
Can_0: 1000.0, 224.0, 224.0, 224.0, 224.0, 224.0, 224.0, 224.0, 224.0, 
Can_1:2000.0, 225.0, 225.0, 225.0, 225.0, 225.0, 225.0, 225.0, 225.0,

so the data are received correctly

stieg, 2016-03-01:
Awesome.  So the issue then lies in the Lua script you are using to parse the message.  Have you looked at the logging output of the Race Capture unit?  There may be clues in there.

Fabrizio82, 2016-03-01:
yes the attached log : 

Can_0: 100.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 
Can_1:200.0, 2.0, 2.0, 2.0, 2.0, 2.0, 2.0, 2.0, 2.0, 
Can_0: 1000.0, 224.0, 224.0, 224.0, 224.0, 224.0, 224.0, 224.0, 224.0, 
Can_1:2000.0, 225.0, 225.0, 225.0, 225.0, 225.0, 225.0, 225.0, 225.0

is from the log on the racecapture. Is still not clear for me what is the ".0" after the id and after each byte data. According to the lua script information i suppose that the interpret see always 11 bit (0) instead of the 29 bit (1)

> rxCAN(channel, [timeout] )
available in firmware 2.0 Receive a CAN message, if available.
params
channel: The CAN channel. 0 for the first channel, 1 for the 2nd, if supported on the hardware
timeout (optional). read timeout in milliseconds. defaults to 100ms
returns
identifier: The Identifier value of the message, either in standard (11 bit) or extended (29 bit) format.
isExtended: 0 for Standard (11 bit) Identifier or 1 for Extended (29 bit) Identifier.
Data: CAN message payload; array up to 8 elements long.
If no CAN message was received, the function returns nil

stieg, 2016-03-01:
> Is still not clear for me what is the ".0" after the id and after each byte data

This is because Lua is representing these values internally as doubles.  It just how Lua works and shouldn't affect your results.

> According to the lua script information i suppose that the interpret see always 11 bit (0) instead of the 29 bit (1)

Not sure what you mean by this exactly.  Would you clarify?

Fabrizio82, 2016-03-01:
ok, i rewrote all the script, now i can receive on both can, 11 and 29 bit . This is a base script for everyone that would like to have can monitoring and virtual channel 
````
-------------------------------------
--------------SETTING----------------
-------------------------------------
--how frequently we poll for CAN messages
tickRate = 30
--the CAN baud rate
CAN_baud = 500000
--CAN channel to listen on. 0=first CAN channel, 1=second
CAN_0 = 0
CAN_1 = 1
--1 for Big Endian (MSB) mode; 0 for Little Endian mode (LSB)
be_mode = 0
--add your virtual channels here
--format addChannel(<name>, <sample rate>, <precision>, <min>, <max>, [units])
CanExternalId_1 = addChannel("EXT_Can_1", 10, 0, 0, 100, "%")
CanExternalId_2 = addChannel("EXT_Can_2", 10, 0, 0, 100, "%")
CanExternalId_3 = addChannel("EXT_Can_3", 10, 0, 0, 100, "%")
CanExternalId_4 = addChannel("EXT_Can_4", 10, 0, 0, 100, "%")
--customize here for CAN channel mapping
--format is: [CAN Id] = function(data) map_chan(<channel id>, data, <CAN offset>, <CAN length>, <multiplier>, <adder>)
CAN_map = {
[100]= function(data) map_chan(CanExternalId_1, data, 0, 1 , 1, 0) end,
[200] = function(data) map_chan(CanExternalId_2, data, 0, 1 , 1, 0) end,
[1000] = function(data) map_chan(CanExternalId_3, data, 0, 1 , 1, 0) end,
[2000] = function(data) map_chan(CanExternalId_4, data, 0, 1 , 1, 0) end
}
-------------------------------------
--------------INIT----------------
-------------------------------------
--Init both can bus
initCAN(CAN_0, CAN_baud)
initCAN(CAN_1, CAN_baud)
--Set tickRate
setTickRate(tickRate)
-------------------------------------
-----------Can Bus LOG---------------
-------------------------------------

function outputCAN()
	repeat 
		id, ext, data = rxCAN(0, 100)
		if id ~= nil then
		print('Can_0: ')
			print(id ..', ')
			for i = 1,#data do
				print(data[i] ..', ')
			end
			println('')
			local map = CAN_map[id]
            if map ~= nil then
                map(data)         
            end
		end
		id, ext, data = rxCAN(1, 100)
		if id ~= nil then
		print('Can_1:')
			print(id ..', ')
			for i = 1,#data do
				print(data[i] ..', ')
			end
			println('')
			local map = CAN_map[id]
            if map ~= nil then
                map(data)         
            end
		end
		
	until id == nil
end

--Map CAN channel, little endian format
function map_chan_le(cid, data, offset, len, mult, add)
    offset = offset + 1
    local value = 0
    local shift = 1
    while len > 0 do
        value = value + (data[offset] * shift)
        shift = shift * 256
        offset = offset + 1
        len = len - 1
    end
    setChannel(cid, (value * mult) + add)
end

--Map CAN channel, big endian format
function map_chan_be(cid, data, offset, len, mult, add)
    offset = offset + 1
    local value = 0
    while len > 0 do
        value = (value * 256) + data[offset]
        offset = offset + 1
        len = len - 1
    end
    setChannel(cid, (value * mult) + add)
end

map_chan = (be_mode == 1) and map_chan_be or map_chan_le

function onTick()

	outputCAN()
	
end
``

Fabrizio82, 2016-03-01:
the issue was related to the processing of the "data" variable and the location of the map channel. now they are on the same "repeat" routine, so they are processed correctly. Another issue that i figure it out is that randomly the app crashes, and show me no connection or a firmware version 0.0.0.0, but this require a different topic

closed 2016-05-05 by stieg