autosportlabs/RaceCapture-Pro_firmware #456: Add getChannel Lua binding

kosenko-max, 2016-02-28:
Currently it's cumbersome to get data from channels - you have to make local variables at the time of update. It would be cleaner code if getChannel would read Virtual Channel data.

stieg, 2016-02-28:
Hey @kosenko-max, Would you provide an example?  I am afraid I'm not quite following you here.  TIA.

kosenko-max, 2016-02-28:
IDC calculation requires to remember IPW in one place and come to RPM in
another place. It can be done by storing last IPW in variables, but that
increases code size and makes it inconsistent with other log data access
methods.

stieg, 2016-02-28:
Got it.  Will look into it.

stieg, 2016-03-07:
New method now [documented on the wiki](https://autosportlabs.net/RaceCapturePro_Lua_Scripting#getChannel.28_.3CChannel_ID_.7C_Channel_Name.3E_.29  Co).  Closing

closed 2016-03-07 by stieg