autosportlabs/RaceCapture-Pro_firmware #464: Further enhancement to RPM filtering

brentpicasso, 2016-03-03:
Based on customer testing feedback:

For 2.8.8
* In RPM only mode
  * Remove dividing duration based on PulsePerRevolution. Rationale: ignition systems attempt a constant dwell time throughout the RPM range, so in some cases we may end the filter too early.  Also, makes it simpler for users to understand the filter length if it is a constant value.
  * Have a sanity filter that ignores RPM values that are 2x the max channel config value. This will filter out any additional spurious signals.

For 2.9.0
Add config values for filter length (uS) and rising / falling edge flags
Have config default be 4.5 ms, falling edge

stieg, 2016-03-03:
> Remove dividing duration based on PulsePerRevolution. Rationale: ignition systems attempt a constant dwell time throughout the RPM range, so in some cases we may end the filter too early.

This doesn't sound right.  Logic:

* On a V8 that red lines at 8K, you get 4 sparks per revolution.
* Assume that you are revving it at 8K
* 4 sparks per revolution means we are seeing a pulse rate equivalent to 32K if there was a single pulse per revolution.  If I didn't scale the quiet period based on this info, their RPM readings would clip once they got above 5K (since the current max is 20K RPM).

The solution here is to use the "max" value set by the user.  Assume the max RPM is there, then we can calculate the absolute minimum value we can use.  Easier to use and sane/logical.  No extra field required.

> Add config values for filter length (uS) and rising / falling edge flags

Happy to add the edge trigger setting.  Filter length I don't want to expose.  Lets use the above calculation?

vtjballeng, 2016-03-04:
Smart coils and ignitors will vary dwell with RPM and noise may not necessarily happen within a fixed filter length window.  Having a variable filter length based on RPM is advisable.  The length should be also divisible based on the pulses per revolution .

Example math.  Let's just say you choose 5ms at 6,000rpm.  Per cylinder, that is 60000/6000 = 10ms, so you are leaving half the period open for a valid signal.  Taking this further and scaling appropriately, at 2,000rpm your filter length would be 60000/2000 = 30ms/2 = 15ms.

If you don't scale this based on the pulses per revolution, you will either drop the signal or allow noise.  

If your sanity filter is a simple cut for 2x the max value, it will only solve a small percentage of the problem.

A filter routine that should deal with both problems is to have a maximum allowable rpm/s delta.  A blip from 1,000rpm to 10,000rpm over 50ms is a 180,000 rpm/s rate of change.  That isn't reasonable for the engines this product is tailored to and would shatter most crankshafts.

It is common to have a filter setting with a wide range accessible to the user on ECUs, loggers, etc (See attached example).  You could choose to combine several filter elements like filter length, maximum value, and delta rpm/s within a single setting that is just a 1-10 value to the customer as an example.

[image]

stieg, 2016-03-04:
Variable is most certainly key.  The algorithm I came up with is as follows:

```c
/**                                                                                                                                                                                                                
 * Calculates the highest quiet period usable based on the timer                                                                                                                                                   
 * configurations.  We calculate this by figuring out the expected                                                                                                                                                 
 * maximum HZ of the engine based in the info provided.  Then we simply                                                                                                                                            
 * take the inverse of that and multiply it by the number of micro seconds                                                                                                                                         
 * in a second.                                                                                                                                                                                                    
 */
static uint16_t get_timer_quiet_period(const TimerConfig *tc)
{
        const float max_rpm = tc->cfg.max;
        const float ppr = (float) tc->pulsePerRevolution;
        const float max_hz = max_rpm * ppr * MAX_RPM_MULT / SEC_IN_A_MIN;

        return max_hz ? US_IN_A_SEC / max_hz : 0;
}
```

This takes the maximum RPM and pulses per revolution when calculating the quiet period (in microseconds).  This should allow us the most flexibility while limiting complexity.

vtjballeng, 2016-03-04:
That is going to be ~80% of the available period.  Example case is 6000rpm again.  Math works out to 8ms max quiet period of the 10ms available at that rpm.  I would make that a 9-10/10 on a filter scale.  I would still incorporate a rpm/s delta as the quiet period will cure some but not all instances.  

The problem with being too aggressive is that you miss the next sample if there is a high rate of change or allow noise and blips if you are too lax on this.  I don't think there is a good way to choose a one size fits all no GUI setting option.  Just ship with 5/10 on the filter scale that has ~4-5ms quiet period at 6000rpm and doesn't allow rpm/s deltas above 20,000 or thereabouts.

stieg, 2016-03-05:
> The problem with being too aggressive is that you miss the next sample if there is a high rate of change or allow noise and blips if you are too lax on this.

Precisely.  Which is why I chose the aforementioned algorithm.  It allows us to be reasonably aggressive while allowing the user to back-off if needed.  The dynamic algorithm also works for people who want to use the RPM reading for things other than engines (like watching the speed of a super-charger or a turbo).

> Just ship with 5/10 on the filter scale

What is this filter scale you speak of?  I see what you wrote above but I don't get it at all.  The numbers seem entirely arbitrary.  If you understand their meaning can you explain it? I have no intention of adding any arbitrary number to the system, so before I would consider it its meaning must make sense.

> So for triggering, you just end up with two variables now. Filter strength (1-10), and edge trigger (rising, falling). 

I will be adding the edge triggering flag so a user will be able to select if they want to trigger on rising or falling edge.

brentpicasso, 2016-03-05:
I think the scale James is proposing is an abstract scale, which would range maybe from "no filtering" to "a lot of filtering".   

Kind of speaks to making it easy for non-technical people to understand, but at the same time, leaving our details raises other issues. 

Possibly, keeping some hard number present - milliseconds of filtering, and also adding easy to understand labels describing the aggressiveness of the filter would be appropriate for the beginner.

no filtering ============||========= max filtering
                                       4.5ms

stieg, 2016-03-05:
> Kind of speaks to making it easy for non-technical people to understand, but at the same time, leaving our details raises other issues. 

If we want to make it easy for non-technical people, we do that in the UI only IMHO.

> Possibly, keeping some hard number present - milliseconds of filtering, and also adding easy to understand labels describing the aggressiveness of the filter would be appropriate for the beginner.
>
> no filtering ============||========= max filtering
> 4.5ms

Seems all very UI to me.  Under the hood I want to ensure that what we expose on the API is just straight engineering logical.  I feel that exposing a "quiet period" time is simply too complicated, given that we would expect users to adjust that in relation to the pulses per revolution value.  That is my primary concern, and hence why I am pushing for they dynamic calculation I implemented above.  That said, I think I will add the following to the timer json: `filtering` where acceptable values are `none`, `auto` (the default), or a number which represents the number of uS of quiet time.  That seems logical and flexible to me.  Thoughts?

vtjballeng, 2016-03-05:
RPM/s delta is still the strongest filter to use imo.

RPM Max is a sensible addition to the other filters.

A quiet period is sensible too but poses more risk than the RPM/s delta for dropping valid data and allowing invalid data.

If you combine all three, you end up with a stronger filter system encompassing a higher percentage of failure modes.  The trick with all three is to use them with mild aggression to have the lowest level of false positives.

As far as the Filter 1-10 scale, exposing something like this is commonplace in industry.  

Let's take the isolated example of the quiet period and let's assume we use the variable system Andrew has provided an algorithm for.  In my prior message, I calculated with his algorithm that the quiet period would be 8ms of a 10ms period @ 6000rpm and said that should be a 9-10 on the scale.  You don't need to expose more than a single simple filter setting.

So, if 80% of the period is a 10/10 on the scale, then you might use the following example set @6000rpm (by just adding a filter factor variable to the existing equation):
Filter value - % of available period @ 6000rpm - corresponding ms quiet period @ 6000rpm
10 - 80% - 8ms @ 6000rpm
9 - 75% - 7.5ms @ 6000rpm 
8 - 70% - 7ms @ 6000rpm
7 - 60% - 6ms @ 6000rpm
6 - 50% - 5ms @ 6000rpm
5 - 40% - 4ms @ 6000rpm
4 - 30% - 3ms @ 6000rpm
3 - 20% - 2ms @ 6000rpm
2 -10% - 1ms @ 6000rpm
1 - 0% - 0ms @ 6000rpm

So, taking this a step further, this is what I think the filter should look like @ 6000rpm (notes, I haven't done the math for the crank speed delta values so all are just examples):

10 - 6ms quiet - crank speed delta < 10000rpm/s + disallow any value at 2xMaxRPM
9 - 5.5ms quiet - crank speed delta < 12500rpm/s + disallow any value at 2xMaxRPM
8 - 5ms quiet - crank speed delta < 15000rpm/s + disallow any value at 2xMaxRPM
7 - 4.5ms quiet - crank speed delta < 17500rpm/s + disallow any value at 2xMaxRPM
6 - 4ms quiet - crank speed delta < 20000rpm/s + disallow any value at 2xMaxRPM
5 - 3.5ms quiet - crank speed delta < 25000rpm/s + disallow any value at 2xMaxRPM
4 - 3ms quiet - crank speed delta < 30000rpm/s + disallow any value at 2xMaxRPM
3 - 2ms quiet - crank speed delta < 35000rpm/s + disallow any value at 2xMaxRPM
2 - 1ms quiet - crank speed delta < 40000rpm/s + disallow any value at 2xMaxRPM
1 - 0ms quiet - crank speed delta < 1000000rpm/s + disallow any value at 2xMaxRPM

If all you did was the rpm/s, this would be best.  If you spend a fraction more time rpm/s + maxRPM is better. rpm/s + maxRPM + quiet period would be best but most complex.

stieg, 2016-03-15:
Merged

closed 2016-03-15 by stieg