autosportlabs/RaceCapture-Pro_firmware #468: Send track id of detected track in telemetry stream

ryandoherty, 2016-03-09:
It would help RCL a lot if RCP sent the track it has detected in the telemetry stream, either upon connecting or as some other type of command/message. 

Potential methods:

- If RCP has already detected the track, include it in the auth command (maybe?)
- When RCP detects the track (post-connect), automatically send a message to RCL with the track id it detected
- Allow RCL to query RCP over the telemetry link for the track.

stieg, 2016-06-05:
For this to work we would want to send it as a separate message.  This way it can happen at any time.  What info do you need for the track exactly?  Id?  Name?

brentpicasso, 2019-02-23:
This requirement should change to a mechanism for Podium to set the active track if RCP doesn't know the track it should use.