autosportlabs/RaceCapture-Pro_firmware #471: Script reload is not removing existing virtual channels

kosenko-max, 2016-03-12:
From time to time change of script would not reset existing virtual channels list, so it hits channel count limit and requires power restart of hardware.

stieg, 2016-03-12:
Yep.  Virtual Channels are only designed for additions.  I'm assuming you are hitting this because you are testing Lua Scripts?

kosenko-max, 2016-03-13:
I assume that firmware and app should clean-up table of virtual channels
after script update. Am I right?

stieg, 2016-03-13:
Not quite.  We expect you not to change the channel name.  hence the name check.  Now if that is broken... Then there is a problem

kosenko-max, 2016-03-13:
Well, since I don't fit in 30 channels I have to jump between scripts and
power restart hardware every time. I just don't really see a point in
storing old channels if script has updated. I can understand a reason to
store app settings of channels even if they don't exist anymore, but not
channels themselves in a firmware or app channel list.

brentpicasso, 2016-03-13:
Good engineering practices encourage separation of concerns, and that would suggest that updating the script would not reach into a different part of the system to do something - touching virtual channels, for example. That may end up having an unintended side effect someone else with a different, equally justifiable use case.

Perhaps a lua function to clear existing channels might help, then it can be at the top of every script, if necessary?

kosenko-max, 2016-03-13:
That would work also, I just can't imagine any scenario where channels must
be persistent after script updates. And since power cycling would clean up
this table anyway, whatever this scenario is - it would not work
consistently anyway.

stieg, 2016-04-11:
in 2.9.0

closed 2016-04-11 by stieg