autosportlabs/RaceCapture-Pro_firmware #476: Re-evaluate Lua GC behavior.  See if we can optimize it further

stieg, 2016-03-17:
Another customer has reported that they were hitting Lua memory issues.  He says he was able to work around it by invoking GC manually at the end of each Lua invocation (by invoking `collectgarbage()`).  Need to investigate why this is the case and see what can be done to prevent it.  Script is attached with user requested modifications (proprietary codes removed)

[Code - For ASL.lua.txt](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/178307/Code.-.For.ASL.lua.txt)

brentpicasso, 2016-03-17:
I think lua has some runtime options that lets you tune GC behavior, would
be worth looking into.

stieg, 2016-03-17:
It does and I have those plumbed into `#define` values already.  What I need to do is lightly profile LUA and see why people are so easily able to exhaust the memory supply.

kosenko-max, 2016-04-02:
This script can't work more than 0.1 second without constant pushing of garbage collector. https://gist.github.com/kosenko-max/c5cb086292c287e45004

It's worth calling garbage collector somewhere here https://github.com/autosportlabs/RaceCapture-Pro_firmware/blob/26ab29d3fd52cfe94112f21bc0310747c9ec754d/src/lua/luaTask.c#L77

kosenko-max, 2016-04-02:
It's also worth looking at defining LUA_OPTIMIZE_MEMORY = 1 (or 2 if to add some functions manually)

I will try to see if adjusting pause or step multiplier would help avoiding manual collection.

stieg, 2016-04-03:
> I will try to see if adjusting pause or step multiplier would help avoiding manual collection.

:+1: This is where I would have started as well.  Can be tricky to get it right without slowing down LUA.

kosenko-max, 2016-04-03:
Well, if out of memory is almost here - it's still good thing to do gc.
Latest Lua does it. Not 5.1.4

stieg, 2016-05-15:
Patch is out and there are very large improvements in there based on the testing I did.  Bottom line... If we want fast LUA we must do everything to avoid constant allocation and discarding of objects.  Getting rid of that can make an order of magnitude difference in run time speed.  Static objects (as one might logically expect) yield the largest performance gains per cycle in our LUA scripting environment.

kosenko-max, 2016-05-15:
Main obstacle is memory. And with that in mind - keeping non local objects
is expensive luxary for code above 5kb.

stieg, 2016-05-15:
> Main obstacle is memory.

Yep.  Its is and always has been.  Kinda like gravity... one of those constants where you are forced to work with it...

> And with that in mind - keeping non local objects is expensive luxary for code above 5kb.

What do you mean by this exactly?  Example?  So long as code doesn't do a lot of allocation of temporary data structures you should be able to do a hell of a lot in Lua in a short amount of time.  Even if you need to use some temporary structures, if you statically allocate them at the global level and re-use them between invocations you will be fine.  Hell even using local variables should be fine so long as you are using primitives and don't abuse the heap.

stieg, 2016-05-16:
Merged

closed 2016-05-16 by stieg