autosportlabs/RaceCapture-Pro_firmware #487: BT initialization requires a long time under certain conditions

brentpicasso, 2016-03-22:
If BT re-initializes while an android device is connected, it will take a very long time vs. if an android device is not connected.  This will considerably impact user experience when experienced. 

r/2.9.0 = 24 seconds from boot to transmitting telemetry
2.8.7 = 4 seconds from boot to transmitting telemetry

If you observe the log immediately after a sysReset you can see it pauses for a long time between baud rates; and also seems to pause for a long time after it falls back to the default baud rate; possibly there might be additional delays injected.

stieg, 2016-03-22:
Yeah... I saw this the other day as well.  Thanks for confirming its not an issue on 2.8.7.  Will dig into this.

stieg, 2016-03-22:
I think the issue we saw was due to the BT falling back to 9600 bps in this case.  That was addressed in a different change, but this further improves the connection by adjusting some parameters and removing some redundancy.

stieg, 2016-03-23:
Merged.

closed 2016-03-23 by stieg