autosportlabs/RaceCapture-Pro_firmware #496: Investigate slow SD card init

stieg, 2016-03-29:
Had a user write in with logs showing slow SD card init times when logging starts.  Need to see if there is a good way to improve this to prevent hiccups in sampling.

Unfortunately don't have permission to publicly share the data just yet... so this is a private ASL only link for now: https://docs.google.com/a/autosportlabs.com/spreadsheets/d/1Vao-xtcFnhNGwnwPO-PpxksxVUw8_k3WdzBqvum8igA/edit?usp=sharing

brentpicasso, 2016-03-29:
worth noting; Upon fresh insert of SD card init time will take longer than a subsequent init. This can be observed visually, even.

Possible option: Make sample buffer larger to hold samples to allow time for SD init.

stieg, 2016-06-05:
Some fixes were made to help this in 2.9.1 to help relieve the scheduler of busy waiting that happens in the SD code.  We should verify that this is not a major issue anymore in 2.10.0.  If it is we should try and isolate why.

stieg, 2016-07-10:
After putting in changes for SDcard IRQ priority and fixing corruption issue this issue seems to have disappeared.  Will reopen it if I see it in the future.

closed 2016-07-10 by stieg