autosportlabs/RaceCapture-Pro_firmware #497: Investigate GPS time 'gaps'

ryandoherty, 2016-03-31:
Sometimes in our log files, the GPS time can jump by a much greater amount than it should. Example: logging sensors at 100HZ, but seeing GPS time jump by 90ms sometimes, then -10ms.

Private spreadsheet here showing a customer's log file data (RC_5 tab) https://docs.google.com/spreadsheets/d/1Vao-xtcFnhNGwnwPO-PpxksxVUw8_k3WdzBqvum8igA/edit#gid=1723100027

stieg, 2016-03-31:
This is likely due to the GPS chip acquiring better data and adjusting its idea of UTC time accordingly.  Need to look into exactly what info we can get from this GPS receiver and how to best use it to ensure we get accurate time values.  May need to introduce a time smoothing algorithm to allow our internal UTC clock to slowly skew towards what the GPS device thinks the real time is.

brentpicasso, 2016-03-31:
One thing I'm curious, are the jumps always forward in time or go backwards?

stieg, 2016-03-31:
Either way.

closed 2018-05-17 by brentpicasso