autosportlabs/RaceCapture-Pro_firmware #498: Investigate GPS speed 'stepping'

ryandoherty, 2016-03-31:
A customer sent us some of their data which showed their GPS speed in a stair-stepped pattern:

[image]

Their throttle was pinned at the max and this behavior was repeated over multiple runs, so most likely not a vehicle specific issue.

stieg, 2016-03-31:
This may be some form of race condition between when snapshots are taken and when data is provided.  The idea being that if the data is late arriving and is provided at the same rate as the snapshot, then we may see a duplicate value of the previous sample, giving the stair effect.  Ideally we could find a way to ensure these sample rates are offset so that data providers internally provide the data somewhere in between the snapshot time, thus eliminating the stepping.

FHB-Performance, 2016-04-01:
Hi Stieg. Thanks for you comment! What do you mean of snapshots are taken? How could I avoid this? Is this related to the fact that I am recording and having the RaceCapture app open at the same time?

FHB-Performance, 2016-04-01:
Especially because I had other tests that I did not have this issue - Different day and different track.

stieg, 2016-04-01:
> What do you mean of snapshots are taken?

Snapshots are the same things as samples.  The idea is that our data sources supply data as quick as they reasonably can and then when the time comes to take a sample a snapshot of the data state is taken.  This data is then converted into the sample form and sent to the various data sinks (like USB, SDcard, Bluetooth, Cellular etc).

> Is this related to the fact that I am recording and having the RaceCapture app open at the same time?

No but there is another thing that could affect your results.  When samples are taken over USB, they are done using a polling method.  Put differently, the app pings the device for a sample, and the race capture device returns a sample.  This happens back and forth quite quickly and could potentially affect results since it adds more load to the CPU.  I am actively working on design changes that will address this behavior in future releases (hopefully 2.10.0, but no promises).

FHB-Performance, 2016-04-03:
Ok Stieg. I will repeat the tests with no app or laptop connected to RCP and see if there is any improvement. 

Thank you again!

stieg, 2016-06-05:
Hey @FHB-Performance did you ever get a chance to repeat this test?  I haven't seen this locally on our end so if you have any further data it would be most useful.  Thanks.

closed 2018-05-17 by brentpicasso