autosportlabs/RaceCapture-Pro_firmware #500: Evaluate heap usage and file tickets to reduce this as much as is reasonably possible.

stieg, 2016-04-05:
It has become quite clear that we are using more RAM that is generally sane/reasonable.  To avoid this and restore stability in the system we need to focus on using less RAM where possible.  This ticket tracks doing a review of the code, identifying large sinks of RAM, and trimming/eliminating them as needed.  Places of note:

* Message Rx Buffers for incoming data
* Serial buffers
* Lua Library usage
* Memory buffers allocated for API commands

This ticket only tracks searching and identifying these problems.  Lets then file a ticket for each one and address accordingly.

brentpicasso, 2016-04-05:
Also should analyze CCM usage vs heap; we may be leaving much memory on the table in CCM where some malloc'd structures can be moved to.

stieg, 2016-04-05:
Very possible.  I can tweak some of our objects to make use of statically allocated buffers if necessary, its just not as ideal.

closed 2018-06-14 by brentpicasso