autosportlabs/RaceCapture-Pro_firmware #503: Ensure LapStats sample rates are consistent at runtime

brentpicasso, 2016-04-08:
lapstats channels have different sample rates, but the app hides it via the UI for simplification. 

There's a chance that lapstat sample rates can get out of sync, and can only be corrected if the config is read and lapstats config is written back - leaving the chance of a bad config lingering.

Firmware could compensate at runtime by selecting and normalizing on the highest configured sample rate for all lapstat channels.  

...this is a band-aid until which we modifiy the API to only have a single common samplerate.

stieg, 2016-04-08:
> There's a chance that lapstat sample rates can get out of sync, and can only be corrected if the config is read and lapstats config is written back - leaving the chance of a bad config lingering.

Do you have any info on how they are getting out of sync?

brentpicasso, 2016-04-10:
Unsure. likely due to an old configuration state not getting updated from a previous firmware upgrade.

closed 2016-07-20 by stieg