autosportlabs/RaceCapture-Pro_firmware #505: Use Sortable DeviceId+DateTime in Log File Names and cleanup old logs

kosenko-max, 2016-04-10:
Currently RCP saves log files with numbers from 0 to 99999. https://github.com/autosportlabs/RaceCapture-Pro_firmware/blob/e680676af72fb258a1c5603d1e79529b5f56ebc4/src/logger/fileWriter.c#L234 Problem with this approach is that it's hard to tell which file has what. Plus if some logs have to be cleaned manually - sequence will be broken.

Proposal is to use sortable date time in the file name according to ISO8601.
So it's 2016-04-10T00-01-12.log. In order to do it more universal and self descriptive - I suggest to use time as GMT+0 and add DeviceId in front of the date, so that log files from several RCP would not mix up. 

In order to do uninterrupted logging when space on SD card is coming to an end - it should delete oldest log file by name in the background. So far I don't see automatic cleanup at all (did I miss something?)

kosenko-max, 2016-04-10:
It's also makes sense to add logic of making new log files every 100Mb otherwise it will fail at 4Gb limit and automatic cleanup would be deleting too much.

stieg, 2016-04-10:
> Proposal is to use sortable date time in the file name according to ISO8601.

Not a bad idea.  As it is now if you look at the file creation time stamp we use UTC time if available.

> It's also makes sense to add logic of making new log files every 100Mb otherwise it will fail at 4Gb limit and automatic cleanup would be deleting too much.

Why every 100MB?  Anything special about that number?  I agree about the 4G value... but I actually haven't heard of anyone hitting that yet in the field.

kosenko-max, 2016-04-10:
Nothing special about 100Mb - it's not too big to analyze it or
automatically cleanup (at 32Gb SD card that means cleaning by 0.3% chunks)
and it's not too small to overload FAT directory index (320 files). Some
analyzers certainly incapable of dealing with 100Mb log files, so it makes
sense to put maximum log size in config.

Right now if you log at 1000hz 100 channels, 8 bytes each - it would reach
4Gb limit in 1.5 hours. I'm not yet aware in details how low frequency data
is updated in the log - is it skipping repetitive values? But 100 virtual
channels is a new thing and I'm coming to the point where I actually will
be able to generate that many. I guess not a lot of folks came to this
point yet.

stieg, 2016-04-10:
>  Some analyzers certainly incapable of dealing with 100Mb log files, so it makes sense to put maximum log size in config.

Valid point.  Some re-tooling of the log file writing is on the road map so I will keep this in mind as that logic is re-tooled.

> I'm not yet aware in details how low frequency data is updated in the log - is it skipping repetitive values?

Nope.  File logging is pretty simplistic and writes every sample value for the sampled channel at that time.  If a channel is sampled at a lower rate, it is skipped.  That is an interesting idea though...

> But 100 virtual channels is a new thing and I'm coming to the point where I actually will be able to generate that many. I guess not a lot of folks came to this point yet.

Heh... yeah you might be a wee bit ahead of the pack here... just a tiny bit...

brentpicasso, 2016-04-10:
The file format is sparse plain text CSV, so channels with slower sample rates are skipped. This was to enhance write performance (to avoid writing extra data) as well as to save space.  You can easily see the effect by setting one channel to 10Hz and another channel at 1Hz and then loading it into a spreasheet.

kosenko-max, 2016-04-10:
> Some analyzers certainly incapable of dealing with 100Mb log files, so it
> makes sense to put maximum log size in config.
>
> Valid point. Some re-tooling of the log file writing is on the road map so
> I will keep this in mind as that logic is re-tooled.
I was hoping to propose you binary log system with extraction of logs
through the app (telemetry, local unpacking, google drive/dropbox storing,
etc). Currently it's anyway not convenient to take out logs, so it makes
sense to store raw integer data in a very compact form (50 times more
compact at least). If you have any ticket for new log writing - I'd like to
propose some algorithms.

It could be tremendously more compact - that gives ability to save
telemetry traffic, get whole set of data in the app through the congested
BT channel without losing any resolution. I was heavily involved in a data
packing systems before. It could be useful. I would propose pull requests,
but my full-time job is factory management, so I have only part of weekend
to do code and my dev environment is still doesn't work. Plus I don't have
any bench device and development in the car is not that comfortable at all.

> I'm not yet aware in details how low frequency data is updated in the log
> - is it skipping repetitive values?
>
> Nope. File logging is pretty simplistic and writes every sample value for
> the sampled [...]
How far away in road map new file logging system?

> But 100 virtual channels is a new thing and I'm coming to the point where
> I actually will be able to generate that many. I guess not a lot of folks
> came to this point yet.
>
> Heh... yeah you might be a wee bit ahead of the pack here... just a tiny
> bit...
I'm on a 3rd edition of channel mapping - I can't keep definitions of can
maps as direct code, can't keep them as a set of arrays, can't read from
external files (Lua io module is not implemented), now I'm trying to read
from very compact 2kb string in the code. If anyone made 100 useful virtual
channels already - I'd like to see how they fit in 50kb Lua memory.

kosenko-max, 2016-04-10:
Right, I just checked wiki about that. It's good, simple avoidance of same
value would add something extra. But it's still would be far away from
refined compact binary protocol. But I've already said about it...

brentpicasso, 2016-04-10:
@kosenko-max it's very important to have a native log file that is easy to process by our customers. The compacted / sparsely populated CSV provides the best balance between ease of use, compactness and performance, so the plan is to keep this format.

kosenko-max, 2016-04-10:
CSV can be easily done at the App or on Podium. Could you share with me
sample use-case when it won't work and it have to be stored on RCP this
way? I'm just curious.

Thanks.

brentpicasso, 2016-04-10:
A very common use case for our customers is that they log data to SD, then pull SD card and load data into their own applications for their own purposes (excel, or other analysis software).

kosenko-max, 2016-04-10:
Ok. As far as I understand they do it, because its the only way to do it
now. Is there access to drive through USB connect?

What I propose opens up possibility to get CSV from Podium (select set of
sessions, click export CSV) or auto sync it to Google Drive through the
phone/tablet App without even taking SD card out or doing anything.

But for those who still prefer to deal with SD cards moving around - just
use same App. Register file extension (RCL or something) and make it auto
export to CSV with run of default CSV registered app on result.

I certainly understand why sparse CSV output is a primary requirement. But
if binary storage allows you to have full scale logs in the cloud
immediately and get out CSV any moment - isn't it just better?

I personally would prefer that App/Podium gives me navigation over whole
set of logs through the history, ability to classify regions and do simple
select region or few with one click export by template to CSV/GEMS and run
associated app for instant analysis. Dealing with SD cards and numbered log
files is quite cumbersome and not very "cloudy".

closed 2018-05-17 by brentpicasso