autosportlabs/RaceCapture-Pro_firmware #516: Document timer range of timer channels

brentpicasso, 2016-04-13:
Need to document lowest measurable frequency / RPM range of timers 1-3 for MK2, for each timer speed (low / medium / high), expressed as practical measurement range. e.g.  x to y Hz

stieg, 2016-05-15:
Each timer buffer counts up to a maximum value of 2^16 (even the 32 bit timer... I adjusted it to make it consistent with the others).  So our Maximum value is 2^16 - 1 = 65535.  Now our clock coming in (on MK2) is a 168MHz clock.  For fast, medium and slow we have 3 pre-scalar values.  These divide the clock by their value.  Here are the current values as of 2.9.1:

Speed | Prescalar
-------- | ------------
Fast | 84
Medium | 168
Slow | 1680

So with those values in mind plus our maximum timer counter value we can calculate the lowest measurable frequency.  The values are as follows:

Speed | Lowest (Hz) | Lowest (RPM) | Highest (Hz) | Highest (RPM)
-------- | ------------- | ----------------- | ------------- | -------------------------
Fast | 30.5 | 1831 | 2000000 Hz | 120000000
Medium | 15.25 | 915 | 1000000Hz | 60000000
Slow | 1.5 | 90 | 100000 Hz | 600000

Based on that info... we should probably recommend that people use the "Slow" setting.  It should more than cover their needs.  Hell even medium isn't that great.  Thoughts @brentpicasso ?

brentpicasso, 2016-05-15:
It's a balance. you want reasonably high accuracy at the upper RPM range, where the pulsewidth is shortest, and you also want to measure down to a low RPM without overflowing. 

Also, you have to consider that most ignition systems will have 2,3 or 4 pulse per revolution, not a single pulse per revolution as what I think you calculated above. So, Medium I think is still the best option.

stieg, 2016-05-15:
Oh yes... I forgot the PPR calculation in that.  Most people have at least 2 PPR so that makes mediums low level 407.  Still closer than what I would prefer... but its not bad.  I think if we doubled or quadrupled medium's value in the next release it would be ideal.

brentpicasso, 2016-05-15:
Most cars do not idle that low, so I think it's OK as is, and preserves accuracy at the upper RPM range, where people are really caring about things.

stieg, 2016-05-15:
Well... most well functioning cars don't idle that low.  Also I'm thinking of corner cases that we would like to support as well.  Think of a two cylinder or one cylinder engine.  Motorcycle, Go-Kart, etc.  Gotta keep those in mind.  And multiplying by a 2 - 4 x is trivial resolution loss.  At a medium value our resolution is 1 micro second.  We could easily live with that value sitting at 2 or 4 micro seconds.

stieg, 2016-05-15:
If we were really clever bastards... we could have the users provide us the highest and lowest ranges they wanted to measure and then calculate a prescalar based on those values.

brentpicasso, 2016-05-15:
https://en.wiktionary.org/wiki/too_clever_by_half

brentpicasso, 2016-05-15:
;)

stieg, 2016-05-18:
Can we close this or should this info go somewhere specific?

brentpicasso, 2016-05-18:
should go in the wiki, not sure where exactly.

stieg, 2016-05-18:
Yeah I'm having a hard time with that one too.  There is also the wiki we
maintain on GitHub for the firmware too.

brentpicasso, 2016-05-18:
I'll find a spot. Probably the hardware installation guide cross referenced
with the software guide

Brent Picasso
Autosport Labs
Technology for Race and Street

stieg, 2016-05-23:
Closing as @brentpicasso noted he would port this to the installation guide.

closed 2016-05-23 by stieg