autosportlabs/RaceCapture-Pro_firmware #522: Add lua scripting, Bt, cellular and wifi capabilities to the getCapabilities command

ryandoherty, 2016-04-20:
Our standardized method of determining if a RC/RCP device can support certain commands or configurations is to query via `getCapabilities`. Currently RC does not support lua scripting or BT and a RCP device may or may not support WiFi/cellular configuration. The RC app needs to know this to determine what configuration screens to show.

The new sections to add to the response to `getCapabilities` should be:

* `lua`
* `bt`
* `wifi` 
* `cellular`

stieg, 2016-05-11:
For `lua` look for the `"script"` variable under `"db"`.  If its there then we support Lua.  Else we do not.  I will add a communications section for you.

stieg, 2016-05-12:
This is now done.

closed 2016-05-12 by stieg