autosportlabs/RaceCapture-Pro_firmware #524: Disable Watchdog when RCP Panics

stieg, 2016-04-24:
If RCP hits one of our real Panic conditions, we should disable the watchdog so that we see it.  Perhaps this only should happen in the non-prod builds, at least until we have full logging to SD so users can send in log files.

Jeff-Ciesielski, 2016-04-25:
On the STM32F3/4, once the IWDG is started, it can't be disabled.  One way I've found to deal with this in prod is to keep an area of ram off limits (much like we do for the bootloader handshake) and store some error information + a stack frame.  This ram is unknown to the startup code, so it doesn't get zeroed out on reset.  If an error is detected upon reset (via a magic value or flag bit set somewhere in that scratch pad), the watchdog is not enabled, and a special panic state is entered which allows dumping the debug data.

stieg, 2016-06-05:
As of this writing we currently do disable watchdog on tagged builds so errors are very obvious.  So there is nothing more to solve there.  However @Jeff-Ciesielski raises a good idea of how to capture an error and log it.  So I am going to hijack this bug to track that purpose going forward.  This now tracks implementing some form which users in the field can report a crash bug to us.  This would better allow us to diagnose and solve crash issues quickly.