autosportlabs/RaceCapture-Pro_firmware #529: Support sparse data when it comes to flashing NVRam

stieg, 2016-04-24:
Currently when we flash NVram we allocate a block of RAM the size of the chunk to be flashed.  This is silly and inefficient and needs to be replaced with a method that will allow us to do a sparse flash where we just provide what we need to flash and provide 0's for the rest of the flash space.

Jeff-Ciesielski, 2016-04-25:
This can get tricky depending on the flash controller.  Sinking a flash cell to 0 means it can't be written again unless the entire page or sector (depending on the controller) is re-charged so you don't want to 'provide zeros for the rest of flash space' since that would lead to large dead spots.  Also, these controllers have a minimum write size (I think it's 16 bits, but I'd have to double check) which is uncommonly small.  So while you can do really small incremental writes, it's better to subdivide into blocks so that you can efficiently keep track of what's been erased unless you're doing something trick like journaling.

TL;DR: Unless I'm fundamentally misunderstanding the problem, performing a block write is generally the correct/safe thing to do with NOR flash.

stieg, 2016-04-25:
I think there may be some misunderstanding here.  What I am referring to is that when we write a new sector of data (say the config), we allocate a whole large block of RAM that is the size of that section (say 16K).  If the config is 10K, then we are needlessly allocating 6K additional that won't be used in any way.  My goal is to only use the 10K, not allocate the 16K block, and have the system handle this sanely.  It may already do this, I just filed this to follow up and check that it does or doesn't.

> performing a block write is generally the correct/safe thing to do with NOR flash.

Agreed.  I have no plans on moving away from this model.

Jeff-Ciesielski, 2016-04-25:
Gotcha, I read 'chunks' and assumed you meant 'blocks'.

stieg, 2016-06-05:
I investigated this further last night.  The only real optimization that can easily be had here would be to create some minimal structs where the binary layout is the same yet the default data is only what is flashed instead of the entire struct.  However since resetting is such an infrequent task it doesn't seem worth the effort.  Time would be better spent ensuring that enough memory is available prior to flashing and finding ways to make that more resilient (like always killing Lua before we flash).  Anyways going to close this as won't fix for now.

closed 2016-06-05 by stieg