autosportlabs/RaceCapture-Pro_firmware #536: v2.9.0: Cannot update RCP track maps if non trivial script is loaded

brentpicasso, 2016-04-26:
If a script is running it is possible that a trackmap cannot be loaded due to insufficient memory. 

STR:
* load a non-trivial script, such as below. 
* add some trackmaps to config.
* observe trackmaps not added upon read-back
* observe error in log console "Failed to allocate memory for track buffer" 

Example script:

--This example adopted to Hondata KPro4 CAN output 

--how frequently we poll for CAN messages, 30Hz is the max 
tickRate = 30 
--the CAN baud rate 
CAN_baud = 250000 
--CAN channel to listen on. 0=first CAN channel, 1=second 
CAN_chan = 0 

--add your virtual channels here 
rpmId = addChannel("RPM", 25, 0, 0, 9000, "RPM") 
vltId = addChannel("EcuVolts", 10, 1, 0, 20, "volts") 
iatId = addChannel("IAT", 1, 0, 0, 100, "C") 
ectId = addChannel("EngineTemp", 1, 0, 0, 150, "C") 
tpsId = addChannel("TPS", 10, 0, 0, 100, "%") 
mapId = addChannel("MAP", 10, 1, 0, 15, "PSI") 
injId = addChannel("InjectorPW", 10, 3, 0, 100, "ms") 
ignId = addChannel("Ignition", 10, 0, -20, 20, "D") 
lmdId = addChannel("AFR", 10, 3, 0, 2, "lambda") 
knkId = addChannel("Knock", 1, 0, 0, 15, "count") 
camId = addChannel("CamTiming", 10, 0, -20, 20, "D") 

function toF(value)
 return value * 1.8 + 32
end

function toAFR(value)
 return value * 14.7
end

--customize here for CAN channel mapping 
--offset/length in bytes? 
--format is: [CAN Id] = function(data) map_chan(<channel id>, data, <CAN offset>, <CAN length>, <multiplier>, <adder>) 
CAN_map = { 
--did not bother logging gear speed and target cam angle 
[1632] = function(data) map_chan(rpmId, data, 0, 2, 1, 0) map_chan(vltId, data, 5, 1, 0.1, 0) end, 
[1633] = function(data) map_chan(iatId, data, 0, 2, 1, 0, toF) map_chan(ectId, data, 2, 2, 1, 0, toF) end, 
[1634] = function(data) map_chan(tpsId, data, 0, 2, 1, 0) map_chan(mapId, data, 2, 2, 0.0145037738, 0) end, 
[1635] = function(data) map_chan(injId, data, 0, 2, 0.001, 0) map_chan(ignId, data, 2, 2, 1, 0) end, 
[1636] = function(data) map_chan(lmdId, data, 0, 2, 0.00003051757, 0, toAFR) end, 
[1637] = function(data) map_chan(knkId, data, 0, 2, 1, 0) end, 
[1638] = function(data) map_chan(camId, data, 2, 2, 1, 0) end 
} 

function onTick() 
 processCAN(CAN_chan)
 --processLogging()   
end 

function processLogging()
 if getGpio(1) < 1 then
   startLogging()
 else
   stopLogging()
 end
end

--===========do not edit below=========== 
function processCAN(chan) 
    repeat 
        local id, e, data = rxCAN(chan) 
        if id ~= nil then 
            local map = CAN_map[id] 
            if map ~= nil then 
                map(data)          
            end 
        end 
    until id == nil 
end 

--Map CAN channel, big endian format 
function map_chan(cid, data, offset, len, mult, add, filter) 
    offset = offset + 1 
    local value = 0 
    while len > 0 do 
        value = (value * 256) + data[offset] 
        offset = offset + 1 
        len = len - 1 
    end 
 local cv = value * mult + add
 if filter ~= nil then cv = filter(cv) end
    setChannel(cid, cv)
end 

initCAN(CAN_chan, CAN_baud) 
setTickRate(tickRate)

stieg, 2016-06-05:
Probably memory contention issue here.  I bet if you were to look at the response code of the API it would be a failure due to lack of memory.  Solution is to stop Lua, update tracks, then restart Lua.

stieg, 2016-06-09:
@brentpicasso I can't seem to reproduce this on 2.9.2.  From my logs I see that we are stopping Lua, loading the tracks and then restarting lua.  This should prevent any and all memory issues.  Are you seeing this on your end?  I would paste my logs but since copy/paste is broken on my system currently I am unable to do so

stieg, 2016-06-09:
Screenshot as proof.  Closing as can't reproduce now in top of tree 2.9.2.  Lets revisit this if there is a way to reproduce it with given files.
[image]

stieg, 2016-06-10:
I hit this myself while using the AIM Smartycam script.  This is on 2.9.2

[image]

stieg, 2016-06-10:
Right... just did some digging to see what our options are.  Frankly, they are extremely limited.  In theory if our devices had reasonable erase page sizes we could get away with modifying a single page at a time rather than copying the whole sector into RAM, modifying it, then storing it again.

On RCT this seems to be an option since flash page sizes are 2K, but on MK2 we are boned since we can only erase an entire Sector!  Yeah... so that eliminates the idea there about read/modify/write on page boundaries.

The only way we could get around this would be to have the app _always_ send all the tracks it wants to save to the DB (in essence, write the whole DB every time you change it).  By doing this I could just write the structure directly into NVRAM and not need to allocate anything.  Need to think about this more.  But this isn't getting fixed for 2.9.2.

stieg, 2016-06-15:
After discussion about future of how tracks will be loaded we have decided that app will always send whole DB when updating track DB.  Therefore we just need to adjust the API to not be sparse going forward.  This will prevent the allocation of the large RAM block which will solve the problem.

stieg, 2016-07-18:
A working solution is available but since https://github.com/autosportlabs/RaceCapture_App/issues/1059 exists it likely means we will need to bump the API version before we put this in place.  Will discuss in tech corner of weekly meeting.

stieg, 2016-07-18:
Bumping from 2.10.0 since it requires an API version bump.

closed 2018-05-17 by brentpicasso