autosportlabs/RaceCapture-Pro_firmware #544: Timeout of rxCAN is slowing down other LUA functions

karstenbreitkopf, 2016-05-01:
When using the rxCAN function it slows down other calculations, depending on the value of the rxCAN timeout parameter. Simple example for RCP scripting.. Integrate the sample time in 1ms to get "real-time"
counter. With rxCAN timout of 100ms the timer "TEST" is slowed down extremely already. With timeout of 0.0 ms it does´n slow down that much.

Example: 
> setTickRate(1000)
> TestId = addChannel( "TEST",100)
> --- END CHANNEL CONFIG ---
>
> --- BEGIN GLOBAL TIMER VARIABLES ---
> timer10MS=0
> timer100MS=0

> --- BEGIN TASK SCHEDULER
> function onTick()
>  --1ms task
>  task1MS()
  
>  --10ms task
>  if timer10MS < 10 then 
>  timer10MS=timer10MS+1
>  else
>  task10MS()
>  timer10MS=0 
>  end
>  --100 ms task
> if timer100MS < 100 then 
>   timer100MS=timer100MS+1
>  else 
>    task100MS()
>    timer100MS=0 
>  end
> end
> --- END TASK SCHEDULER ---
>
> --- BEGIN 1MS FUNCTION CALLS ---
> function task1MS()
> integrate_time()
> end
> --- BEGIN 1MS FUNCTION CALLS ---

> --- BEGIN 10MS FUNCTION CALLS ---
> function task10MS()
> CAN_receive_10MS()
> end
> --- END 10MS FUNCTION CALLS ---
> function task100MS()
>
>end
>--- END 10MS FUNCTION CALLS ---

> function CAN_receive_10MS()
>    id, ext, data = rxCAN(0,100)
>    if id ~= nil then
>    if id == 534 then -- ENGINE_1 (0x216)
> 
>    end  
> end
> end

test = 0
test_prev = 0
function integrate_time()
test = test_prev + 0.001
test_prev = test
setChannel(TestId, test)
end

brentpicasso, 2016-05-01:
This is by design, as the onTick() function runs all calls synchronously.

 If you want fast response in other parts of the script, we recommend a minimal timeout and then handle the case if no message is received (return value is nil)

closed 2016-05-01 by brentpicasso

kosenko-max, 2016-05-01:
It can be solved by async methods that could spend a time waiting for
answer or delivery.
Or by more flexible CAN mapping. Or by making separate thread for CAN
messaging.

One example - I'm trying to make fuel pumps management. And this is just
not working properly with CAN communication that can stuck for a long time.
So either I collect logs or use fuel pumps...

stieg, 2016-05-02:
> It can be solved by async methods that could spend a time waiting for answer or delivery.

Problem with async is that depending on the CAN traffic you could get a lot of unnecessary calls.  What we need is a way to register what CAN PIDs to listen to and then when you run the LUA script you always get the latest value of that PID.  I believe issue #172 will help solve this since we will probably need to implement something like that to get it to work.

> One example - I'm trying to make fuel pumps management. And this is just not working properly with CAN communication that can stuck for a long time.

Why is the CAN getting stuck for a long time?  Are you doing things in a synchronous fashion.  With the 1000Hz `onTick` support in 2.9.0 you should be able to set a timeout value of 1ms, and if you don't get it do the other stuff quick and move on.  Or are you trying to read several messages?

kosenko-max, 2016-05-02:
Just don't forget that some CAN bus require request-response and bits extraction before doing channel math and then async would not be needed.

I see what you mean here - I will try this approach, thank you.

It would probably even work better if you allow 0 CAN timeout, so that if CAN message is not in the buffer already it just returns immediately (or if transmit queue is full). I would need to rewrite my poor CAN messaging pipeline again. Don't know how soon you will do CAN mapping with request-response, but I'm fighting with low memory and rewrote code 4 times already.

kosenko-max, 2016-05-02:
Oh, I see that 0 timeout should not stall anything. Thanks for good tip.

stieg, 2016-05-02:
> P.S. I think that it makes sense to add to documentation how to do nonblocking messaging since CAN/Serial messaging is in fact async already, but it's not clear without code digging.

Added a note about that in the parameter description for that method on our wiki: https://www.autosportlabs.net/RaceCapturePro_Lua_Scripting#rxCAN.28channel.2C_.5Btimeout.5D_.29